riscv-boom/riscv-boom #11: the "BOOMCPPConfig" and "BOOMVLSIConfig" can't be recognized

jia1546, 2016-06-21:
When I execute the command "make verilog CONFIG=BOOMVLSIConfig" in the fsim sub-dircectory, the error occurs: "Unable to find configClassName "BOOMVLSIConfig", did you misspell it? " And the similar error occurs when execute command "make run CONFIG=BOOMCPPConfig" in the emulator sub-directory.

ccelio, 2016-06-21:
What is the commit hash of your rocket-chip repository? Are you on the boom branch, 62e9cc5fb5715d95f03b5782fc68bec0dc982336 (the latest boom branch of rocket-chip)?

If that commit hash doesn't work, give (a33c7a51d9ed98a1307c0ee18d711f02a8768977) a try. 

Both of those hashes are of rocket-chip (https://github.com/ucb-bar/rocket-chip/tree/boom).

ccelio, 2016-07-04:
I am unable to replicate your error when following the instructions laid out in the README (https://github.com/ucb-bar/riscv-boom).

closed 2016-07-04 by ccelio