riscv-boom/riscv-boom #13: MegaBOOMConfig fails when checking requirement

sashimi-yzh, 2016-09-29:
I try to use the `MegaBOOMConfig` in the `boom` branch of the `rocket-chip` repo, but it fails at the following requirement in `rocket/frontend.scala:126`
```
require(fetchWidth * coreInstBytes <= rowBytes && isPow2(fetchWidth))
```
where `fetchWidth = 4, coreInstBytes = 4, rowBytes = 8`

ccelio, 2016-09-29:
Huh, I'm surprised rowBytes is now 8.  We'll have to trace that back and see where that's coming from and if we can change it when using 4-wide BOOM.

ccelio, 2016-10-02:
Thanks to @aswaterman , the temporary workaround is to change a parameter in coreplex/Configs.scala:

````bash
boom-chip/src/main/scala/coreplex$git diff Configs.scala 
diff --git a/src/main/scala/coreplex/Configs.scala b/src/main/scala/coreplex/Configs.scala
index 6e00c3f..840ebb1 100644
--- a/src/main/scala/coreplex/Configs.scala
+++ b/src/main/scala/coreplex/Configs.scala
@@ -27,7 +27,8 @@ class BaseCoreplexConfig extends Config (
   topDefinitions = { (pname,site,here) => 
     type PF = PartialFunction[Any,Any]
     def findBy(sname:Any):Any = here[PF](site[Any](sname))(pname)
-    lazy val innerDataBits = site(XLen)
+//    lazy val innerDataBits = site(XLen)
+    lazy val innerDataBits = 128
     lazy val innerDataBeats = (8 * site(CacheBlockBytes)) / innerDataBits
     pname match {
       //Memory Parameters
````

I think I'm going to leave this "fix" uncommitted for now (but issue still open), as a longer-term solution will fall out of the updating of the rocketchip tilelink, which will hopefully alleviate these sorts of couplings.

Let me know if anything else shows up when using 4-wide BOOM.

sashimi-yzh, 2016-10-03:
After this modification, the code passes the requirement checking. To make the implementation fit in the FPGA, I modify `ROB_ENTRIES` and `PHYS_REGISTERS` for the `MegaBOOMConfig`.
```
case "ROB_ENTRIES" => 64
case "PHYS_REGISTERS" => 96
```

But unfortunately, I can not boot linux with `MegaBOOMConfig` with the modifications above. I just get some garbage output
```
[ i08e[    0.n[ m[00[ bo[0mg[0:00             vvv rrvvv rrrE[    0.000000] Li4io 120[ 0e
```
I can boot linux with the same bbl and linux payload by using `MediumBOOMConfig`. So I guess there may be something wrong with the BOOM code.

ccelio, 2016-10-03:
Hmmm, there's three possibilities:

* 1) changing the `innerDataBits` value exposed a bug in the memory system.  

* 2) there's an existing bug(s) in the Rocket-chip memory system that is exposed by wider/higher performance BOOM.

* 3) there's a bug in MegaBOOM that is not exposed by smaller BOOMs.

I have a few questions for you to help us narrow down the problem:

 * When you ran MediumBOOMConfig, did you use the new `innerDataBits=128`?  If so, we can probably eliminate option 1.  

* Can you try modifying MediumBOOMConfig to use a much smaller ROB, and then separately, a much smaller Physical Register File, and verify it still works? It's possible that bugs could be exposed by making those data structures much smaller than normal.

* Can you run MegaBOOM in Verilator simulation and see if trying to boot Linux throws an assertion? That'd be a huge help. The smaller of a failing program you can find, the better.

colinschmidt, 2016-10-03:
For what its worth, Hwacha uses not exactly innerDataBits but sets the number of beats on the inner network to 4 (from default 8) which ends up creating a 128bit interface, just like setting innerDataBits. You should still verify but just wanted to let you know its not entirely untested.

sashimi-yzh, 2016-10-04:
Here are some trials:
* `MediumBOOM` with `innerDataBits=128` -> fail
* `MegaBOOM` with `innerDataBits=128` -> fail
* `rocket` with `innerDataBits=128` -> pass
* `MediumBOOM` with `innerDataBits=site(XLen)` -> pass
* `MegaBOOM` with `innerDataBits=site(XLen)` -> fail
* `rocket` with `innerDataBits=site(XLen)` -> pass

where `pass` means linux is successfully booted on FPGA, and `fail` means not.

I try Verilator with riscv-test by running
```
make run-asm-tests
```
and
```
make run-bmark-tests
```
But all outputs just show the number of cycles they run:
```
Completed after 233485 cycles
```
And I can not tell whether they finished successfully. Can I conclude that everything goes well if nothing wrong is reported?

Finally, I try Verilator to boot linux with `MegaBOOM` and `innerDataBits=128`. First I modify bbl to let `mcall_console_putchar()` return immediately. Then I start Verilator by
```
./emulator-TestHarness-BOOMConfig +max-cycles=1000000 +verbose bbl.elf
```
It reports timeout without any other information after several minutes.

ccelio, 2016-10-04:
Thanks for looking further into this.

The `make run-asm-tests` will complain nosily if there is a failure. They use bash return codes to signal to Make if an error occurs, so if everything is successful the command quietly returns 0.  You can also verify by looking at the output/*.out files and verifying that they are empty (errors will be written to them).

    ./emulator-TestHarness-BOOMConfig +max-cycles=1000000 +verbose bbl.elf

Remove the "max-cycles" command, that is causing the timeout- booting and running Linux is a very long process! Much more than a million cycles.

sashimi-yzh, 2016-10-05:
This time I run Verilator to boot linux without the `max-cycles` argument. After about 9.5 hours, it finish without any outputs, and the command just returns `0`. Anyway, no assertion is thrown.

ccelio, 2016-10-05:
Yah, I'm seeing the same thing on my end; I can't get MegaBOOM to fail in Verilator, and I've been running riscv-torture and booting Linux as test cases for the past two days.

It's possible that there is a bug in the FPGA infrastructure. We are in the process of updating how we interface between rocket-chip and the zynq FPGAs, which should move traffic off of the debug port and onto, I believe, a serial device.

I've updated BOOM to the latest rocket-chip (which in 2-3 weeks had gotten 350 commits behind!), but the fpga-zynq repo hasn't been updated yet to match it.  Hopefully, we'll be done in the next few days and we can give the updated BOOM/FPGA infrastructure another try.

ccelio, 2016-10-29:
@sashimi-yzh we've updated the fpga-zynq repo to use a Tether Serial Interface, and I've been noticing better reliability. Try and give it a go with MegaBOOM.

sashimi-yzh, 2016-10-30:
Actually, I do not have a zynq board. I use vc709 board. And I use a microblaze core to transfer images to the memory of rocketchip, and connect a uartlite directly to rocketchip. So the updated fpga-zynq repo may not help me too much, and I do not need the front end server, either. :)

I have do some trials with the updated boom repo. This time MediumBOOM with `innerDataBits=128` can boot linux successfully! MegaBOOM with `innerDataBits=128` still fails to boot linux. But I can see valid output, instead of garbage output I saw last time. Because I do not use the fpga-zynq repo, I may conclude that the garbage output bug is fixed by some of the commits of the boom repo. :)

This time MegaBOOM fails to boot linux because of random crash like
```
[    0.000000] Memory: 2036688K/2076672K available (1410K kernel code, 83K rwdata, 316K rodata, 6340K init, 187K bss, 39984K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.010000] pid_max: default: 32768 minimum: 301
[    0.010000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.020000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.050000] devtmpfs: initialized
[    0.050000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.060000] clocksource: Switched to clocksource riscv_clocksource
[    0.080000] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
[    0.090000] Oops [#1]
[    0.090000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.6.2-g37ff849 #192
[    0.090000] task: fffffffffc654000 ti: fffffffffc652000 task.ti: fffffffffc652000
[    0.090000] sepc: ffffffff806c276c ra : ffffffff806c28dc sp : fffffffffc653b18
[    0.090000]  gp : ffffffff807f5c60 tp : fffffffffc654000 t0 : 0000000000000006
[    0.090000]  t1 : 0000000000000000 t2 : 0000000000000001 s0 : fffffffffc201480
[    0.090000]  s1 : 0000000000000000 a0 : fffffffffc201480 a1 : 0000000000000000
[    0.090000]  a2 : fffffffffc201500 a3 : fffffffffc201520 a4 : 000000000a00080f
[    0.090000]  a5 : 0000000000000000 a6 : fffffffffc201540 a7 : 0000000000000000
[    0.090000]  s2 : fffffffffc62c800 s3 : fffffffffc60c000 s4 : fffffffffffff000
[    0.090000]  s5 : 0000000000000000 s6 : fffffffffc6a81ca s7 : ffffffffffffff8c
[    0.090000]  s8 : 0000000000000022 s9 : 0000000000008000 s10: 0000000000000001
[    0.090000]  s11: 0000000000004000 t3 : 0000000000000000 t4 : 0000000000000000
[    0.090000]  t5 : 0000000000000009 t6 : fffffffffc2014bb
[    0.090000] sstatus: 0000000000040120 sbadaddr: 0000000000000008 scause: 0000000000000007
[    0.190000] ---[ end trace 2e764554a55f7705 ]---
[    0.200000] Kernel panic - not syncing: Fatal exception
[    0.200000] ---[ end Kernel panic - not syncing: Fatal exception
```
Following is the disassembly result for `vmlinux` near `sepc`:
```
172256 ffffffff806c2754:   00042783            lw  a5,0(s0)
172257 ffffffff806c2758:   08000737            lui a4,0x8000
172258 ffffffff806c275c:   00e7e7b3            or  a5,a5,a4
172259 ffffffff806c2760:   00f42023            sw  a5,0(s0)
172260 ffffffff806c2764:   01813083            ld  ra,24(sp)
172261 ffffffff806c2768:   01013403            ld  s0,16(sp)
172262 ffffffff806c276c:   00813483            ld  s1,8(sp)
172263 ffffffff806c2770:   02010113            addi    sp,sp,32
172264 ffffffff806c2774:   00008067            ret
172265 ffffffff806c2778:   807f57b7            lui a5,0x807f5
172266 ffffffff806c277c:   40678793            addi    a5,a5,1030 # ffffffff807f5406 <__warned.29600>
172267 ffffffff806c2780:   0087c703            lbu a4,8(a5)
```
From the saved registers, I can not figure out why the sbadaddr is `0000000000000008` at `sepc`, since `sp` is not zero.

What makes it more tricky is that this kind of crash occurs randomly with different `sepc`, `sbadaddr`, and `scause`. I have observed `scause=4, 5, 6, 7`. But `4` means `Reserved`. This may exposes that `scause` is untrusted. Sometime the crash happens after unpacking initramfs, and sometime after loading busybox.

ccelio, 2016-10-30:
> Actually, I do not have a zynq board. I use vc709 board.

Ah, that's actually helpful. That hopefully eliminates FPGA infrastructure as a common variable.

> I have do some trials with the updated boom repo [...] I may conclude that the garbage output bug is fixed by some of the commits of the boom repo. :)

Glad to hear we're making progress.  There's been a lot of churn in chisel3, firrtl, and rocketchip's uncore, so there's a lot of moving parts here (and I expect a ton of churn for at least the next month or two). We did find and fix two reset bugs that only manifested when using the new tether serial interface, but was masked away when using the debug module for program loading and core reset. -.-

Unfortunately, I'm still seeing issues with 4-wide BOOM on the FPGA that I can't replicate in simulation.  We have a couple of ideas at Berkeley to help us debug this, but it will take us a while to get them up and running (e.g., making the Chisel asserts synthesizable).  In the meantime, keep me posted.

ccelio, 2016-11-19:
Just to give a heads up, this issue may be resolved by this  bugfix to rocket-chip:

https://github.com/ucb-bar/rocket-chip/pull/441

You can cherry pick that for now,  It'll probably be ~2-3 weeks before I get around to re-evaluating this myself.

sashimi-yzh, 2016-11-30:
I spent several days in fixing a silly bug. I forgot to change the default spinning bootrom. :(

The good news is that, after applying the patch above, MegaBOOM works well!

ccelio, 2016-11-30:
Awesome! I'm anticipating that I'll do a rocket-chip master merge late December/early January so it may be a few weeks before I pull these changes in.

ccelio, 2017-03-20:
I've updated BOOM* and reproduced your error. I'll try to find time to actually look into it.

https://github.com/ucb-bar/rocket-chip/tree/boom-devel
https://github.com/ucb-bar/riscv-boom/tree/devel

*No hacks required to build MegaBoom now. However, Verilator chokes hard on it, so I'm using VCS.

ccelio, 2017-04-09:
This has now been addressed by the latest merge from the `devel` branch.

closed 2017-04-09 by ccelio