riscv-boom/riscv-boom #17: BOOM lacks efficient WFI support

ccelio, 2016-11-06:
Legally, WFI (wait for interrupt) can be handled as a NOP. However, BOOM should strive to be more efficient in how it handles WFI.

As a starting point, this signal from the CSR file is used to provide WFI support in Rocket (`csr.io.csr_stall`) by informing the Decode stage to stall:

https://github.com/ucb-bar/rocket-chip/blob/boom/src/main/scala/rocket/csr.scala#L506

ccelio, 2017-10-20:
This has been addressed by https://github.com/ucb-bar/riscv-boom/commit/98c60153c6c49e15aeb5664d702755dd6295fc7d on the devel branch.  Once that is merged to master, we can close this.

poweihuang17, 2017-10-25:
@ccelio  Could I ask you about the WFI implementation that you added in 98c6015? Did you use a bit from CSR to let the pipeline stall when a WFI is executed?

ccelio, 2017-10-25:
Yes, the WFI is given to the CSRFile, since that is where external interrupts are managed. The CSRFile then asserts a `csr_stall` signal until it something happens. That `csr_stall` signal is then sent to the ROB and commit is blocked.

https://github.com/ucb-bar/riscv-boom/commit/98c60153c6c49e15aeb5664d702755dd6295fc7d#diff-69f876ff27fd9890b973a8d95cadfc0eR480

poweihuang17, 2017-10-25:
@ccelio Thank you for your quick reply and help! We will cover that  in our recent bi-weekly news. Hope to attract more user come to BOOM : )

ccelio, 2018-03-12:
This is now a part of the master branch.

closed 2018-03-12 by ccelio