riscv-boom/riscv-boom #18: Mismatch between short history and long history with no branch predictor?

donggyukim, 2016-11-20:
I got an assertion fail when I ran rv64ui-p-simple even though I turned off the branch predictor:

```
 Assertion failed: [brpredictor] mistmatch between short history and very long history implementations.
     at brpredictor.scala:192 assert (r_ghistory_commit_copy === vlh_commit,
```

ccelio, 2016-11-20:
What's the boom hash you're using? I may have made updates and I may not have bumped the boom branch of rocket-chip to match. 

Worst case, you can freely comment out that assert.

ccelio, 2016-11-20:
I just pushed this, which hopefully won't give you a problem: https://github.com/ucb-bar/rocket-chip/commit/2fcf18cb0b8de39389a9598cfcda4280eff30a3f

donggyukim, 2016-11-20:
I tried it with 4825d37ceef10a329ccaef84bb4247c6e8f34750 with serial port in testchipip. I couldn't try it with the latest commit with it due to coreplex type conflicts, but the failure doesn't appear with the latest commit in rocket-chip. Thus, I close it.

closed 2016-11-20 by donggyukim

ccelio, 2016-11-20:
Ah, you can use the latest boom commit with an older rocket-chip, you just need to apply `git revert` to it, to revert this commit (https://github.com/ucb-bar/riscv-boom/commit/3cdd4bd84891c7b5704f21ff2249ea2ba79a0f14) that updates boom to the latest lazyModule stuff.  You can check out my fpga repo on A5:/nscratch/celio/fpga-zynq/rocket-chip to see what versions I'm using together.

donggyukim, 2016-11-20:
Your tip also works for me and I created a branch reverting that commit. Thanks, Chris!