riscv-boom/riscv-boom #20: Cannot boot linux with newer chisel

donggyukim, 2017-02-26:
It's very odd, but BOOM doesn't boot linux with chisel newer than 4d684398a1774a8b6a7a86e9d2dd92a165e02cfe. It seems like new APIs break some logic in BOOM.

ccelio, 2017-02-26:
So it can run (https://github.com/ucb-bar/chisel3/commit/4d684398a1774a8b6a7a86e9d2dd92a165e02cfe), but it's broken on the next commit (https://github.com/ucb-bar/chisel3/commit/09282f32bec847bcd0b652f778f42be13c7d027e)?

That looks like a scary rabbit to look into.

ccelio, 2017-02-26:
Does rocket boot Linux successfully?

ccelio, 2017-02-26:
And does this manifest in Verilator and/or VCS, or are you trying it on the FPGA?

ccelio, 2017-02-26:
Do you have early printk enabled in the Linux kernel?

linux-4.6.2/.config:
````
#
# Kernel hacking
#
CONFIG_CMDLINE_BOOL=y
CONFIG_CMDLINE="debug earlyprintk norandmaps"
CONFIG_CMDLINE_OVERRIDE=y
CONFIG_EARLY_PRINTK=y

````

The Logo is printed by the riscv-pk loader thingy, and the Linux printfs will get buffered until much later. Turning on early printk will give you better visibility. 

But if this doesn't manifest in VCS, this is gonna be a really hard thing for me to debug. Especially if it's a Chisel bug dealing with bulk connects/clonetypes.

donggyukim, 2017-02-26:
Yeah, I think that's the case. Eventually, you may want to turn off the compatibility mode.

ccelio, 2017-02-26:
Did you try this in VCS and it booted fine? 

Have you tried running a simpler, bare-metal program on FPGA boom?

If "boom-chip/vsim" worked, have you tried the VCS simulation setup provided in the `fpga-zynq` repo? (https://github.com/ucb-bar/fpga-zynq/tree/master/simulation)

It's possible that the use of the Debug module is hiding an issue that only manifests using the TSI. Wouldn't be the first time a boom bug only manifested from the use of the TSI (the previous was a resetVector bug that was hidden by the DTM).

donggyukim, 2017-02-28:
FYI, chisel@224ee397db324067b35874bf22c76a63e9ca531b breaks boom. The main difference is bulk connections are expanded in chisel, which was done in firrtl before.

ucbjrl, 2017-02-28:
Can someone try ucb-bar/chisel3#530 and see if this resolves the issue?

donggyukim, 2017-03-01:
fixed by ucb-bar/chisel3#530

donggyukim, 2017-03-01:
It still doesn't boot yet while runs pk hello

donggyukim, 2017-03-02:
This is resolved with ucb-bar/chisel3#530 and #21.

closed 2017-03-02 by donggyukim