riscv-boom/riscv-boom #22: MegaBOOMConfig error

owahidcse99, 2017-03-13:
Hi,
I can build BOOMConfig and MediumBOOMConfig in emulator directory  but when I used make CONFIG=MegaBOOMConfig in emulator dir, I get the exact same "java.lang.reflect.InvocationTargetException in run-main" error as the following link
https://groups.google.com/forum/#!topic/riscv-boom/KOyNci70kYM

Looking into the link https://github.com/ucb-bar/riscv-boom/issues/13
I changed rocket-chip/src/main/scala/coreplex/Configs.scala
   -  lazy val innerDataBits = site(XLen)
   + lazy val innerDataBits = 128
And I also changed, rocket-chip/boom/src/main/scala/configs.scala
class WithMegaBOOMs extends Config(
   knobValues = {
..................
case "ROB_ENTRIES" => 64
case "PHYS_REGISTERS" => 96
Default were 128 for both. Any other combination is giving me error during MegaBoom build.

Now that the MegaBOOMConfig build has passed, I am getting error in running elf files.

Assertion failed: [BROB] Committing an entry with no executed branches or jalrs.
    at brpredictor.scala:651 assert (entries_ctrl(head_ptr).debug_executed === Bool(true),
[810249] %Error: rocketchip.MegaBOOMConfig.v:929329: Assertion failed in TOP.TestHarness.ExampleTop_1.coreplex.BOOMTile_1.core.bpd_stage.br_predictor.brob
%Error: /home/owahid/boom-rocket/rocket-chip/emulator/generated-src/rocketchip.MegaBOOMConfig.v:929329: Verilog $stop
Aborting...

I commented line 651 in boom/src/main/scala/brpredictor.scala and did the config again. Then I got second assertion as following.

Assertion failed: [brpredictor] VLHR: commit bit doesn't match speculative bit.
    at brpredictor.scala:482 assert (debug_com_bit  === taken, "[brpredictor] VLHR: commit bit doesn't match speculative bit.")
[810255] %Error: rocketchip.MegaBOOMConfig.v:931690: Assertion failed in TOP.TestHarness.ExampleTop_1.coreplex.BOOMTile_1.core.bpd_stage.br_predictor
%Error: /home/owahid/boom-rocket/rocket-chip/emulator/generated-src/rocketchip.MegaBOOMConfig.v:931690: Verilog $stop
Aborting...

I commented line 482 again and now getting the 3rd assertion as following which I did not commented out so far.

Assertion failed: [brpredictor] mistmatch between short history and very long history implementations.
    at brpredictor.scala:192 assert (r_ghistory_commit_copy === vlh_commit,
[810256] %Error: rocketchip.MegaBOOMConfig.v:931590: Assertion failed in TOP.TestHarness.ExampleTop_1.coreplex.BOOMTile_1.core.bpd_stage.br_predictor
%Error: /home/owahid/boom-rocket/rocket-chip/emulator/generated-src/rocketchip.MegaBOOMConfig.v:931590: Verilog $stop
Aborting...

I was wondering is it possible to have MegaBOOMConfig from a clean repo without going through these hacks?

Thanks,
owahid

ccelio, 2017-03-13:
What binary are you running? Is this something that I can easily replicate on my end?

owahidcse99, 2017-03-13:
It is simple hello program elf file. I used ./emu*MegaBoom* pk hello.elf

ccelio, 2017-03-13:
Okay, a few comments:

* I'm in the process of updating BOOM to the latest rocket-chip tilelink protocol. I expect to be done in the next two weeks on this. This will likely change what "hacks" are used to make MegaBOOM work, which is why I've been reluctant to commit them while things are still in flux.

* The assertions you see failing are unfortunate and indicative of a bug that shows up in MegaBOOM.  But they are also in the branch prediction unit, so they shouldn't affect the correctness of BOOM, just it's performance.

* Once I've updated BOOM to rocket-chip's latest stuff, I'll reevaluate MegaBOOM and see if I can reproduce your assertion failures.

ccelio, 2017-03-20:
I've updated BOOM* and reproduced your error. I'll try to find time to actually look into it.

https://github.com/ucb-bar/rocket-chip/tree/boom-devel
https://github.com/ucb-bar/riscv-boom/tree/devel

*No hacks required to build MegaBoom now. However, Verilator chokes hard on it, so I'm using VCS.

ccelio, 2017-04-03:
@owahidcse99 You can give the boom-devel branch a try now, I believe I have addressed the assertions you were seeing. 

You may need to change the configuration of `MegaBoomConfig` to something lower to get it to build. There have been a lot of changes in rocket-chip land which have affected the compilation that we'll need to look into.

Once I'm happy, I'll merge devel branch into master.

commit e9ae170d (2017-04-03): [bpd] fix collision of branch update and commit.

   - Now delay commit signal in BROB to match the already-delayed branch update
     signal (otherwise commit data would be incorrect).
   - Delay flush signal in BROB to match the now-delayed commit signal.
   - Delay flush signal in BPD to match BROB's now-delayed commit signal.
   - These updates were caught by assertions regarding history integrity.

Closes #22.
commit 4857b601 (2017-04-06): [bpd] fix collision of branch update and commit.

   - Now delay commit signal in BROB to match the already-delayed branch update
     signal (otherwise commit data would be incorrect).
   - Delay flush signal in BROB to match the now-delayed commit signal.
   - Delay flush signal in BPD to match BROB's now-delayed commit signal.
   - These updates were caught by assertions regarding history integrity.

Closes #22.
closed 2017-04-09 by ccelio