riscv-boom/riscv-boom #27: verilator got segment falt when generate  MegaBoomConfig cpp files

jphong1, 2017-05-16:
Hi,
I build MegaBoomConfig/MediumBoomConfig and SmallBoomConfig emulators for performance test.
It work well when i build and run Mediu and Small Boom emulator , but when i try to build  MegaBoom,
i got a segment fault when verilator  try to generate cpp source.

here is my build flow
   $ export RISCV=/path/to/riscv/toolchain
   $ git clone https://github.com/ucb-bar/rocket-chip.git
   $ cd rocket-chip
   $ git checkout boom
   $ git submodule update --init
   $ cd riscv-tools
   $ git submodule update --init --recursive
   $ ./build.sh
   $ cd ../emulator; make CONFIG=MegaBoomConfig

and my repo log 
commit 41fe0d86cef4d91f5f62c26a149ef161c324a4b1
Author: Christopher Celio <celio@eecs.berkeley.edu>
Date:   Sun Apr 30 02:33:46 2017 -0700

any suggestions?
BRs,

ccelio, 2017-05-18:
I would need to see more info on exactly what the output was when verilator failed. However, the likely answer is that your machine ran out of memory.

jphong1, 2017-05-22:
Hi,
Make log is below, and i build it on virtualbox 64bits ubuntu 16.04 with 8GB memory.
As you see no more information to know what happened.
But good news is it can  pass if using verilator 3.902 or reduce robentry numbers.

mkdir -p /home/guest01/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/guest01/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig.fir -o /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig.v -X verilog
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
mkdir -p /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig
/home/guest01/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/guest01/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/jimmyhong/riscv-toolchain/include -I/home/guest01/rocket-chip/csrc -DVERILATOR -include /home/guest01/rocket-chip/csrc/verilator.h" -Mdir /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig \
-o /home/guest01/rocket-chip/emulator/emulator-rocketchip-MegaBoomConfig /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig.v /home/guest01/rocket-chip/csrc/emulator.cc /home/guest01/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/jimmyhong/riscv-toolchain/lib -Wl,-rpath,/home/jimmyhong/riscv-toolchain/lib -L/home/guest01/rocket-chip/emulator -lfesvr -lpthread" \
-CFLAGS "-I/home/guest01/rocket-chip/emulator/generated-src -include /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig/VTestHarness.h"
Segmentation fault (core dumped)
%Error: Command Failed /home/guest01/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/guest01/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/jimmyhong/riscv-toolchain/include -I/home/guest01/rocket-chip/csrc -DVERILATOR -include /home/guest01/rocket-chip/csrc/verilator.h' -Mdir /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig -o /home/guest01/rocket-chip/emulator/emulator-rocketchip-MegaBoomConfig /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig.v /home/guest01/rocket-chip/csrc/emulator.cc /home/guest01/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/jimmyhong/riscv-toolchain/lib -Wl,-rpath,/home/jimmyhong/riscv-toolchain/lib -L/home/guest01/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/guest01/rocket-chip/emulator/generated-src -include /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig/VTestHarness.h'
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-MegaBoomConfig' failed
make: *** [emulator-rocketchip-MegaBoomConfig] Error 139

ccelio, 2017-05-22:
> Segmentation fault (core dumped)

Looks like Out-of-Memory to me. You could try and run the `verilator` command in gdb and see if that gives you any additional insight, but probably not. :(

> /home/guest01/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=$c("verbose","&&","done_reset") +define+STOP_COND=$c("done_reset") --assert [...]
-o /home/guest01/rocket-chip/emulator/emulator-rocketchip-MegaBoomConfig /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig.v /home/guest01/rocket-chip/csrc/emulator.cc /home/guest01/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/jimmyhong/riscv-toolchain/lib -Wl,-rpath,/home/jimmyhong/riscv-toolchain/lib -L/home/guest01/rocket-chip/emulator -lfesvr -lpthread" 
-CFLAGS "-I/home/guest01/rocket-chip/emulator/generated-src -include /home/guest01/rocket-chip/emulator/generated-src/rocketchip.MegaBoomConfig/VTestHarness.h"

closed 2017-07-13 by jphong1

owahidcse99, 2017-08-11:
Hi Jphong1/Chris,
Could you please share how the issue was resolved? I am having same segfault issue. Mine is not a virtual machine but Ubuntu on laptop that has enough memory. make debug CONFIG=MegaBoomConfig/MediumBoomConfig gets segfault but make debug CONFIG=BOOMConfig in emulator directory passes.
Thanks,
owahid