riscv-boom/riscv-boom #35: MegaBoomConfig in developement branch: Exception

VaginAY, 2017-10-20:
Good day!

  Here is log of "make verilog CONFIG=MegaBoomConfig"
  Boom branch : devel
  Rocket branch : boom-firesim

> [info] [0.004] Elaborating design...
Interrupt map (2 harts 2 interrupts):
  [1, 2] => ExampleRocketSystem

> WARNING: gen in new Queue(gen, ...) must be a Chisel type, not hardware
WARNING: gen in new Queue(gen, ...) must be a Chisel type, not hardware
WARNING: gen in new Queue(gen, ...) must be a Chisel type, not hardware
WARNING: gen in new Queue(gen, ...) must be a Chisel type, not hardware
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:50)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at chisel3.core.Module$.do_apply(Module.scala:45)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:238)
	at chisel3.Driver$.elaborate(Driver.scala:92)
Caused by: java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:212)
	at freechips.rocketchip.rocket.ICacheModule.<init>(ICache.scala:220)
	at freechips.rocketchip.rocket.ICache.module$lzycompute(ICache.scala:47)
	at freechips.rocketchip.rocket.ICache.module(ICache.scala:47)
	at freechips.rocketchip.rocket.ICache.module(ICache.scala:46)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike$$anonfun$4$$anonfun$apply$9.apply(LazyModule.scala:155)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike$$anonfun$4$$anonfun$apply$9.apply(LazyModule.scala:155)
	at chisel3.core.Module$.do_apply(Module.scala:45)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike$$anonfun$4.apply(LazyModule.scala:155)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike$$anonfun$4.apply(LazyModule.scala:153)
	at scala.collection.immutable.List.flatMap(List.scala:338)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike$class.instantiate(LazyModule.scala:153)
	at freechips.rocketchip.diplomacy.LazyModuleImp.instantiate(LazyModule.scala:175)
	at freechips.rocketchip.diplomacy.LazyModuleImp.<init>(LazyModule.scala:176)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45$$anonfun$apply$46.apply(Defaults.scala:793)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45$$anonfun$apply$46.apply(Defaults.scala:793)
	at scala.Option.foreach(Option.scala:236)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45.apply(Defaults.scala:793)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45.apply(Defaults.scala:791)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 8 s, completed Oct 20, 2017 9:57:30 AM
/home/vagin-ay/Downloads/boom/rocket/vsim/Makefrag-verilog:12: recipe for target '/home/vagin-ay/Downloads/boom/rocket/vsim/generated-src/freechips.rocketchip.system.MegaBoomConfig.fir' failed
make: *** [/home/vagin-ay/Downloads/boom/rocket/vsim/generated-src/freechips.rocketchip.system.MegaBoomConfig.fir] Error 1

Best regards.

VaginAY, 2017-10-20:
MediumBoomConfig compiles good

VaginAY, 2017-10-21:
Identical fail in
  Latest developement commit: 46aef9afae84fa3e4c908e33816734783e5bfee0
  Boom branch : devel
  Rocket branch : boom-devel.

Failing requirement:
  require(tl_out.d.bits.data.getWidth % wordBits == 0)
string 220, 
file rocket-chip/src/main/scala/rocket/ICache.scala.

After its commented out - verilog generation works, but i think it is not clear.

val wordBits = outer.icacheParams.fetchBytes*8
fetchBytes is constant 4, so wordBits=32

About tl_out, i don't understand this scala constructions for now:
val (tl_out, edge_out) = outer.masterNode.out(0)
val masterNode = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
    sourceId = IdRange(0, 1 + icacheParams.prefetch.toInt),
    name = s"Core ${hartid} ICache")))))

ccelio, 2017-10-21:
Thanks for poking around on this (and posting your branches). You can add Scala println to learn more about what the values are of `tl_out` and `wordBits` at Scala runtime/elaboration time.

I believe the issue is there is a mismatch between the TileLink beat-width and the underlying I$ SRAM bank size (`wordBits` I think). We will have to figure out which parameter controls that width and increase it to a size that will work with the MegaBOOM fetch width.

I'm doing quite a bit of refactoring of the code base's organization, so this is not going to get my full attention for a few weeks.

pentin-as, 2017-10-24:
I'm trying to deal with this problem.

Really, in the 220th line in ICache.scala the condition isn't fulfilled:
`require (wordBits % tl_out.d.bits.data.getWidth == 0)`
tl_out.d.bits.data.getWidth = 64
wordBits = 128
Really, the code which is below the condition won't work if it is not fulfilled.

wordBits is set in the 196th line:
`val wordBits = outer.icacheParams.fetchBytes*8`
fetchBytes for MegaBoom = 4*4=16. Thus, wordBits = 128.

The capacity for tl_in is also equal to 128. It is set here:
```
  val slaveNode =
    TLManagerNode (icacheParams.itimAddr.toSeq.map { itimAddr =&gt; TLManagerPortParameters (
      Seq (TLManagerParameters (
        address = Seq (AddressSet (itimAddr, size-1)),
        resources = device.reg ("mem"),
        regionType = RegionType.UNCACHED,
        executable = true,
        supportsPutFull = TransferSizes (1, wordBytes),
        supportsPutPartial = TransferSizes (1, wordBytes),
        supportsGet = TransferSizes (1, wordBytes),
        fifoId = Some(0))),//requests handled in FIFO order
      beatBytes = wordBytes,  <-- here !!!
      minLatency = 1) })
```

But... the capacity for tl_out isn't set. Also it can't be set as tl_out is:
```
val masterNode = TLClientNode (Seq (TLClientPortParameters (Seq (TLClientParameters (
    sourceId = IdRange (0, 1 + icacheParams.prefetch.toInt),//0=refill, 1=hint
    name = s "$ Core { hartid } ICache")))))  

```
... and capacity is possible to set only in TLManagerPortParameters.
We can see it here:
```
  def apply (client: TLClientPortParameters, manager: TLManagerPortParameters) =
    new TLBundleParameters (
      addressBits = log2Up (manager.maxAddress + 1),
      dataBits = manager.beatBytes * 8,  <-- here !!!
      sourceBits = log2Up(client.endSourceId),
      sinkBits = log2Up(manager.endSinkId),
      sizeBits = log2Up (log2Ceil (max (client.maxTransfer, manager.maxTransfer)) +1))
```

As a result, ICache.masterNode connects to TLXbar where too it isn't visible that something was adjusted.

I will continue to deal with this problem.

colinschmidt, 2017-10-24:
Its pretty easy to change the L1->L2 network beat width, and can be done in a much more robust fashion than above:

```
// Set TL network to 128bits wide
case SystemBusParams => up(SystemBusParams, site).copy(beatBytes = 16)
```

ccelio, 2017-10-25:
Fixed by (https://github.com/ucb-bar/riscv-boom/commit/6605572ab7b804f69188bcae1c69337ca3cd1bae). 

Thanks everybody for your help.

commit 6605572a (2017-10-25): [config] Widen Systembus to 16 bytes for BOOM-4w.

Fixes assert error with a mismatch between TileLink nodes as the I$
needs 16 byte fetch width and the SystemBus must be at least as big.

Closes #35.
closed 2017-10-25 by ccelio