riscv-boom/riscv-boom #36: Add chicken-bit support to BOOM

ccelio, 2017-10-20:
**Summary:**
A nice-to-have feature is the ability to enable and disable hardware features from the software.

**Examples:**
* Enable TSO memory consistency
* Enable watchdog timer
* Change predictor behaviors

**High-level thoughts:**
* Do **not** use non-standard Control/Status Registers (CSRs). The address space will get too congested, not indexable, messy to assemble for, etc.
* Instead provide a tile control port MMIO device. Rocket-chip currently uses one to report ECC and bus errors.
* It would be nice to provide a systematic interface so that BOOM control signals can register themselves as part of the chicken-bit system.

ccelio, 2019-01-27:
The new chisel3.util.experimental.BoringUtils may help here.

akshataj96, 2022-03-10:
Hi, Is this still available to pick up?

jerryz123, 2022-03-10:
Yes, we would be happy to accept PRs adding new chicken bits to BOOM.

akshataj96, 2022-03-11:
Okay great! I will start working on it and revert if I have any questions.