riscv-boom/riscv-boom #37: riscv-tests do not work

VaginAY, 2017-10-24:
riscv-tests - last commit in master branch

Test commands:
   cd emulator
   make run CONFIG=MediumBoomConfig  // MegaBoomConfig still under understanding issue#35

Test combinations:
  Boom branch : none, use rocket core
  Rocket branch : master.
  Result: all good.

   Latest developement commit : 46aef9afae84fa3e4c908e33816734783e5bfee0
   Boom branch : devel
   Rocket branch : boom-devel.
   Result: endless execution on first test.

   Boom tag : v2.0.1
   Rocket branch : boom
   Result: fail on first test.

   Boom tag : v2.0
   Rocket branch : boom
   Result: fail on first test.

   Boom tag : v1.0
   Rocket branch : boom-v1
   Result: fail on first test.

  I think all riscv-tests must work for BOOM too, because ISA is similar for all cores.
  It's right? If so we will start debug with Cadence Incisive tomorrow, else please give
  your comments.

  Thanks. Best regards,
  Vagin Alexander

ccelio, 2017-10-24:
There are two issues here.

First, you **must** use the riscv-tests that is found in `rocket-chip/riscv-tools/riscv-tests`. You can **not** use the master branch of riscv-tests. As one example, there was a privileged spec update from v2.0.1 to boom-devel so you must rebuild all of your riscv-tools on a BOOM update.

Second, on `devel` going forward, you **must** build a new compiler and specifically disable the Compressed Extension, since riscv64imafd is now the default compiler option. The `devel` readme describes how (https://github.com/ucb-bar/riscv-boom/tree/devel) under the "Installing the RISC-V Toolchain".

VaginAY, 2017-10-24:
Thanks for fast reply. I'll try it this evening.

VaginAY, 2017-10-24:
Latest developement commit : 3eb9560e93b6491cac9e4b1fc7fba499c02ba340
Boom branch : devel
Rocket branch : boom-devel.
Riscv-tools : commit from boom-devel branch of rocket without any checkouts,
compiled with --with-arch=rv64imafd.
Riscv-tests compile with recompiled riscv-tools.

Commands:
make run-bmark-tests CONFIG=MediumBoomConfig
make run-asm-tests CONFIG=MediumBoomConfig

Result: endless execution on tests
  benchmarks/mm.riscv
  rv64um-v-mul

I'll try v2.0.1 tag tomorrow (and may be v2.0 and v1.0, if it doesn't work).

ccelio, 2017-10-24:
I have reproduced the error on my end.

**Working Config:**
SmallBoomConfig

**BrokenConfigs:**
BOOMConfig
MediumBoomConfig
SmallBoomConfig with fWidth=2

Looks like I dropped the ball on testing superscalar fetch somehow.

ccelio, 2017-10-25:
Fixed. Requires pull of boom `devel` branch and rocket-chip `boom-devel` branch.

commit eebc0dda (2017-10-25): Fix superscalar decode/CSR interaction.

Requires updating rocket-chip.

Closes #37.
closed 2017-10-25 by ccelio