riscv-boom/riscv-boom #38: JTAGVPI/simple JTAG emulator (verilator/VCS/irun)

VaginAY, 2017-10-27:
Good day!

  Order and conditions
  1. Adding new config WithJtagDTM in boom_configs.scala
`
class jtagMegaBoomConfig extends Config(new WithMegaBooms ++ new DefaultBoomConfig ++ new WithNBigCores(1) ++ new WithoutTLMonitors ++ new BaseConfig ++ new WithJtagDTM)
`
  2. Generate verilog.
  cd vsim
  make verilog CONFIG=jtagMegaBoomConfig
  All good.

  3. Try to gen emulator with verilator (as for SimDTM).
    cd emulator
    make run CONFIG=jtagMegaBoomConfig

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586: Cannot find file containing module: JTAGVPI

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586: Looked in:

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/vsrc/JTAGVPI

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/vsrc/JTAGVPI.sv

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       JTAGVPI

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       JTAGVPI.v

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       JTAGVPI.sv

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig/JTAGVPI

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig/JTAGVPI.v

%Error: ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig.v:760586:       ~/Downloads/boom/rocket-chip/emulator/generated-src/freechips.rocketchip.system.jtagMegaBoomConfig/JTAGVPI.sv

  4. After simple renaming vsrc/jtag_vpi.v to vsrc/JTAGVPI.v

%Error: /home/vagin-ay/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v:95: Unsupported: Verilog 1995 reserved word not implemented: wait
%Error: /home/vagin-ay/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v:96: syntax error, unexpected if
%Error: /home/vagin-ay/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v:115: Unsupported or unknown PLI call: $check_for_command
%Error: /home/vagin-ay/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v:142: Unsupported or unknown PLI call: $send_result_to_server
%Error: /home/vagin-ay/Downloads/boom/rocket-chip/vsrc/JTAGVPI.v:151: Unsupported or unknown PLI call: $send_result_to_server

  5. Synopsys VCS usage found for JTAGVPI in makefiles .
  6. A large number of different header files are required for assembly. ?Verilator? headers too!

  Try to run alternative simulation:
  7. Remove JTAGVPI subsystem in TestHarness.v, replace it by simple systemverilog JTAG control module. Make simple clk and rst gen.
  8. Run with cadence incisive (irun) simulation.
  9. BoomCore's program counter is 0x10050 (bootrom) when reset is hi.
      BoomCore's program counter is 0xXXXXX after reset is becoming low.

So, issues:
  1. Is there a method to run BoomCore+JTAGVPI with verilator?
  2. Is there a method to run BoomCore+JTAGVPI with another tools, without VCS?
  3. is clk and reset signals enough to start BoomCore from bootrom? 

Best regards,
Vagin Alexander.

ccelio, 2017-10-27:
Unfortunately, I do not have much knowledge about using different interfaces like SimDTM, Serial or JTAG. 

Have you been able to replicate this effort using rocket-chip? If not, I recommend you start there first. You can also get better help on sifive forums if you run into issues.

My concern would be that you might need a different emulator.cc if you are using a different interface. For example, Berkeley has https://github.com/ucb-bar/testchipip, which provides a SimSerial interface which provides a verilator-harness.cc to replace emulator.cc. It also provides its own bootrom too.

VaginAY, 2017-10-28:
From https://www.veripool.org
Verilator only supports synthesis constructs not testbenches, this is described under "Intro" and in the manual. 

So "wait" construction in jtag_vpi.v is not supported, and PLI calls from jtag_vpi.c too. Verilator is dead branch for JTAG running.

 I will continue to deal with this problem by using cadence tools.

VaginAY, 2017-10-28:
Some additional results of my work.

  1. I try to use results of rocket chip development and merge
   rocket-chip/master branch to rocket-chip/boom-devel.
  2. Only one merge  conflict in file rocket/CSR.scala.
Just use the file from master branch.
  3. Make this changes for compilation:

src/main/scala/decode.scala
-   val csr_decode = Flipped(new freechips.rocketchip.rocket.CSRDecodeInfo)
+   val csr_decode = Flipped(new freechips.rocketchip.rocket.CSRDecodeIO)

src/main/scala/lsu.scala
-   dtlb.io.req.bits.instruction := false.B
 
src/main/scala/parameters.scala
-   val nPerfCounters    = rocketParams.nPerfCounters
+   override val nPerfCounters    = rocketParams.nPerfCounters

  4.cd emulator 
make run CONFIG=MegaBoomConfig
All tests passed.

  But issue with JTAG is still under investigation...

  Many positive changes have been made in master branch of rocket-chip.
Maybe i shoud make pull request with this results?

ccelio, 2017-10-28:
Thanks for the diff list, I've pushed an update to rocket-chip:boom-devel and boom:devel to bring it up to the latest rocket-chip:master.

https://github.com/freechipsproject/rocket-chip/commit/f03ab95deed4e41bffa252de34fd9fdb93cda6e5
https://github.com/ucb-bar/riscv-boom/commit/98bae1f6b066d491c6e07e181e9bef3e73587ff6

FYI, I am currently working on a reorganization of the rocket-chip/boom relationship. My goal is to match the https://github.com/ucb-bar/project-template style in which a "boom-template" master repository includes rocket-chip:master and boom repositories as submodules. This will remove relying on a branch of rocket-chip and better match the other existing rocket-chip based repositories (e.g., sifive's freedom fpga repository).

I don't have an ETA though since it requires moving quite a bit of rocket-chip functionality into the boom repository.

VaginAY, 2017-10-29:
Ok, I did it. But the results are strange.

At first, JTAG modules and "++ new WithJtagDTM" config modification does not affect the operation of the core. The problem of launching the core (from bootrom) turned out to be associated with randomization.

Secondly, I managed to launch my own testbench with JTAG simulation and running core by adding next options to irun (cadence incisive):
  +define+RANDOMIZE                \
  +define+RANDOMIZE_MEM_INIT       \
  +define+RANDOMIZE_REG_INIT       \
  +define+RANDOMIZE_GARBAGE_ASSIGN \
  +define+RANDOMIZE_INVALID_ASSIGN
These definitions are missing from the code because I performed the splitting of the generated file into parts (Verilog::EditFiles from Verilog-perl).

Now I can write my own tests through JTAG interface. However, it seems to me there is another problem or maybe new issue.

In my opinion, randomization is not clear for hardware simulation. Unknown value 'X' covers much more possible situations than randomization. The fact that the core does not work with the use of unknown values ​​theoretically can mean the presence of possible states from which the core will not start.

I think, this issue may be closed. But what about randomization problem? Shoud I create new issue for its investigation?

P.S. About project-template - it's great! However, I believe that it can take a long time.
Maybe we can help you with routine tasks such as a trial branch merging and results testing?
For example, last freechipsproject/rocket-chip@5ca04a5c411f86b6acef4def3fcadeedce7cb1be
can be merged with boom-devel without any conflicts. All emulator tests passed on MegaBoomConfig.

ccelio, 2017-10-29:
In Chisel/rocket-chip, the philosophy is to avoid X's due to issues of X-pessimism. Randomization is our way around X's.

This thread on X-pessimism in rocket-chip may be relevant: https://groups.google.com/a/groups.riscv.org/forum/?utm_medium=email&utm_source=footer#!msg/hw-dev/R32grbbLMLU/PUKwaPV3AgAJ

So it may or may not be useful to track down where X's are being problematic.

I'm hoping the project-template work will take me another week. My hope is that it will make it easier to build BOOM in different environments (new FPGAs, bootroms, devices, etc.). So once I get a prototype out there, I think valuable input will be code review, cleaning up the build system, and trying to add new options (e.g., SimDTM vs SimSerial vs JTAG).

VaginAY, 2017-10-30:
Ok, strange results again.

Here is sch, generated from BTBsa.v:
[image]

Here is verilog BTBsa.v fragments:
`assign s0_idx = last_idx;`
`always @(posedge clock) begin`
`  last_idx <= s0_idx;`

Here is scala btb-sa.scala source:
`   val stall = !io.req.valid && !io.icmiss`
`   val s0_idx = Wire(UInt(width=idx_sz))`
`   val last_idx = RegNext(s0_idx)`
`   val new_idx = getIdx(io.req.bits.addr)`
`   s0_idx := Mux(stall, last_idx, new_idx)`

"stall" and "new_idx" are absent in the generated verilog code.
If I make this modification in verilog:
`  last_idx <= 6'h00;`
BOOMCore starts without randomization, and code execution starts from bootrom.

I do not understand why chisel3 performed such a deep optimization. I will continue to
deal with it tomorrow.

ccelio, 2017-10-30:
I recently made a change to the BTB behavior, pushing the prediction back a cycle. It's possible I made a mistake, or that I left in some vestigial code: https://github.com/ucb-bar/riscv-boom/commit/4898fe845988037d0a3c8d6d820460a2944ab92c

ccelio, 2017-10-30:
Hmmmm, it looks like on the merge to rocket-chip 30 days ago, and the move of the BTB back a cycle,  `icmiss` and `req` are no longer connected.  So we should hardwire `stall` to false.B.

VaginAY, 2017-10-30:
In this case operations with idx do not make sense. I'm right?

ccelio, 2017-10-30:
You're right -- the current BTB is completely and utterly broken. This is why this is on the `devel` branch and not yet `master` ;)

The BTB previously relied on some very invasive changes to the rocket-chip:master's FrontEnd in order to get access to the S0_addr and icmiss signals. I decided in the recent rocket-chip merges (~1 month ago) that I wanted to address this BTB/I$ interfacing issue in another manner.  The BTB is now in an unfinished state and I forgot to wire-off the dangling inputs. I'll push a temporary fix, but it will be a few weeks before I get to finishing the BTB in the manner I want to.

VaginAY, 2017-11-05:
Ok, i ran riscv-tests on my behavioral model. But without scoreboard for now.
Is there a simple method of "tests passed" control? Compare dump with reference or check
any signal/bus state for example... Now I'm investigating the source code of the riscv-tests, but this is not a quick thing.

ccelio, 2017-11-05:
The tests provide a success code to bash. If they fail the fesvr I believe will print out a failure message to stdout. The tests are self-checking so the core tells fesvr when it finishes with the exit code.

ccelio, 2017-11-06:
FYI, here's my new boom-template repo. It's unstable, but you can give it a try: https://github.com/ccelio/boom-template

VaginAY, 2017-11-06:
A little history, maybe it will be useful to someone.
1. As I said before, I use modified config - jtagMegaBoomConfig (++ new WithJtagDTM).
2. Because our team does not use the synopsys software, we can not use the vcs tools for debugging.
Therefore, the JTAGVPI module was removed, and the JTAG interface ports are output to
the outside and connected to a self-developed driver.
3. After several attempts to reach the system bus via jtag, we realized that debug draft specification 0.13 is not fully supported. In particular, there is no access to the system bus. And JTAG driver must do
much more than simple read/write operations.
4. The following algorithm was used to run riscv-tests:
   - use riscv-uknown-elf-objcopy with "-O verilog" option for converting test executable file to verilog $readmemh supported format;
   - with $readmemh and some transformations, the program is configured to load into boom_test.dut.SimAXIMem.AXI4RAM.mem.mem_ext.ram variable;
   - the bootrom was modified to go to the address 80000000 (10040, hang branch);
Analysis of diagrams shows that the test is being executed. Now, the task is to set up auto-monitoring.
To solve this problem, criteria for the positive completion of riscv-test are required.

As follows from the above, I do not use fesvr and C environment. Only generated verilog and riscv-tests executable files. As close as possible to the real system :))
As I understand for now riscv-tests use RVTEST_FAIL and RVTEST_PASS macro for generate result. I'll try to find these macro execution on diagrams and check them for regs or memory cell modification.

P.S. I tried to work with boom-template. Run make in verisim. 
[error] ~/boom/boom-template/src/main/scala/example/Configs.scala:7: not found: object testchipip
[error] import testchipip._
We should write it from scratch or take from the project-template?
Sorry, but I can't create issue in boom-template repo.

ccelio, 2017-11-06:
Thanks for the heads up, I've added Issues and Pull Requests to the `boom-template` repo. Looks like I need to debug some things that show up when building from a fresh repo.

ccelio, 2017-11-06:
With the new https://github.com/ccelio/boom-template setup, you can perhaps try to mimic the sifive freedom platform (https://github.com/sifive/freedom) which should build a self-hosting real system. 

I believe the External Debug spec allows for some things to be left unimplemented assuming you provide other access mechanisms (program buffer vs system bus access, for example). I believe there are some CSRs that inform the system of what exactly is available. I may be wrong though -- none of that is my code.

The RVTEST_FAIL/PASS are writes to a particular, cache-coherent memory address. Some agent (fesvr usually) will need to poll that "tohost" memory address and wait for a non-zero value.

ccelio, 2017-11-06:
I have removed the vestigial `src/main/scala/example` code. Hopefully things work now in boom-template.

closed 2019-12-03 by abejgonzalez