riscv-boom/riscv-boom #39: incorrect mepc

pentin-as, 2017-10-31:
When executing the "ebreak" instruction, the mepc register is set to "PC + 4" instead of "PC".

ccelio, 2017-10-31:
Verified to pass on rocket, fail on boom.

pentin-as, 2017-11-01:
[image]
[image]

When executing an illegal instruction (0x00000000) csr.io.pc is set in the correct state until its value is set in the epc register.
When executing the break instruction csr.io.pc is set in the correct state only one tact, the next value csr.io.pc is set in the epc register.

csr.io.pc is assigned from rob.io.com_xcpt.bits.pc
I think the error should be fixed in rob.io.com_xcpt.bits.pc

ccelio, 2017-11-01:
What PC is your illegal instruction at? What value are you expecting?

ccelio, 2017-11-01:
I think this is the error:

https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/rob.scala#L645

It assumes that any `com_xcpt.bits.pc` that is not an exception wants PC+4.

pentin-as, 2017-11-01:
I expect the PC to be 0x1000000C.

pentin-as, 2017-11-01:
I understand correctly that need a line:
 val refetch_inst = exception_thrown
change to:
 val refetch_inst = exception_thrown | insn_break
(and insn_break should be taken from the decoder)?
Or is it better to try completely to delete the expression "+ Mux(refetch_inst, UInt(0), UInt(4))" and run the tests?

I'll try the second method.

pentin-as, 2017-11-01:
The second method did not help. I will understand how to do it right.

ccelio, 2017-11-01:
Okay, here's how `ebreak` works in BOOM currently.

`ebreak` is a system instruction that is sent to the CSRFile for handling (it goes through the issue window, down the pipeline, and hits the CSRFile during "write back"). It is not an exception, so handling it in the ROB doesn't quite feel right. Trying to set some `insn_break` signal is non-obvious, because commit is looking across a row of potential instructions, some of which may not be valid. It would have to look at the first valid instruction across the potential commit row to see if it is a breakpoint (which can be set by decode).

I'm going to see if I can instead implement `ebreak` by getting its PC in the same manner as a branch instruction and feeding it directly to the CSR file once the `ebreak` flows down the pipeline to set the `csr.io.cmd`.

ccelio, 2017-11-02:
Fixed by https://github.com/ucb-bar/riscv-boom/commit/71146913dc041362490893b413bae38dc10fd5f7. 

This issue will be closed once `devel` is merged into master.

commit 71146913 (2017-11-02): Fix breakpoint setting EPC correctly.

   * Add control signal bit to uop to denote "is_break".
   * If breakpoint is at the head of the ROB, set xcpt.pc to current pc.
   * Breakpoint is a unique instruction, so it will be at the ROB head
      once the instruction hits the CSRFile.
   * A different solution is to read out the breakpoint PC during register-read,
      like a branch instruction, and push it into the CSRFile as the breakpoint
      comes down the pipeline.

Closes #39
closed 2018-03-12 by ccelio