riscv-boom/riscv-boom #40: RenameStage+IssueUnitCollasping X-pessimism.

VaginAY, 2017-11-06:
In these objects there is a loop, which includes uninitializable registers.

r_valids register from rename.scala and io.dis_readys register from issue_ageordered.scala are interdependent. In addition, the r_valids register depends on itself.

This leads to the fact that when randomization is turned off, the execution of this code hangs:
`  li s0, DRAM_BASE`
`  csrr a0, mhartid`
`  la a1, _dtb`
`  jr s0`

After add sync reset to the verilog code to r_valids register (with name _T_****_* after verilog generation), all passed well.

Is this scala correction clear?
```
--- a/src/main/scala/rename.scala
+++ b/src/main/scala/rename.scala
@@ -250,8 +250,11 @@ class RenameStage(
          require (renameLatency == 2)
          val r_valids = Reg(Vec.fill(pl_width) {Bool(false)})
          val r_uops   = Reg(Vec(pl_width, new MicroOp()))
-
-         when (io.kill)
+         when (reset.toBool)
+         {
+            r_valids(w) := Bool(false)
+         }
+         .elsewhen (io.kill)
          {
             r_valids(w) := Bool(false)
          }

```

I believe that this issue can lead to problems at the stage of simulating gatelevel netlist
and should be corrected.

ccelio, 2017-11-08:
Good find; this code should have -- but failed -- to intialize `r_valids`. I'll push a fix.

ccelio, 2017-11-08:
This has been fixed in the `template` branch.

commit 9d34962e (2017-11-08): Fix initialization bug in Rename.

Closes #40
closed 2018-03-12 by ccelio