riscv-boom/riscv-boom #41: GSharePredictor pattern tables preset

VaginAY, 2017-11-09:
Good day!

  Have you any plans about presets for GSharePredictor
pattern tables? If I correctly understood information about
its internal structure, it looks like some kind of simple neural
network. For such systems pre-prepared weights may 
reduce the percentage of mispredictions.
  This will give an additional effect - if you turn off randomization,
the branch predictor will continue to work, and will not fall into
x-pessimism.

Best regards,
Vagin Alexander.

ccelio, 2017-11-09:
The implementation does sort of look like it's reading weight tables, hehe, but I assure you it is not a perceptron-based predictor.

The gshare predictor is conceptually a table of two-bit saturating counters. Gshare is implemented in BOOM using three single-port SRAMs. You can't provide initial values to SRAMs. 

The predictor will very quickly learn (or fail to learn) whatever program it is running -- there is on the order of 100 thousand counters in total -- so there is also no value in providing an initial sets of counter values for gshare even if you could and they will get overwritten with something very quickly.

X-pessimism is an issue, and I would be interested in your thoughts on how to defend the output of the predictor against it. However, the branch predictor is a classic "it's okay to predict garbage" module, so it's way too expensive and needless to provide extra hardware that tracks valid bits or handles memory initialization.

VaginAY, 2017-11-13:
One additional question.
// In btb-sa.scala

Suppose that the index and the corresponding tag match, thanks to random content of btb_tag_array.
Is there a problem with incorrect random data from btb_data_array which will be read at the same index? Will the system fix it?

ccelio, 2017-11-13:
There is a checker module that verifies the integrity of predictions --- that is, that a prediction was performed on something that is a branch (or jump) and that the predicted target matches what could actually happen (https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/fetch.scala#L629-L635).

VaginAY, 2017-11-14:
Thank you for answer.
Only one idea with init state of GSharePredictor for now. Here is its logic.
1. In set associative branch target buffer add nWays number of vectors with name tags_init. Width of these vectors is nSets. One vector for each btb_tag_array.
2. For each vector create initial state ({nSets{1'b0}} or {nSets{1'b1}}) with external reset.
3. When a branch instruction is detected, check the state of the corresponding bit.
4. If the state of corresponding is initial, this means that the corresponding memory cells are garbage. Else memory cells prepared before.
5. If the state of corresponding is initial, all rams outputs of branch predictor system should be ignored or replaced by defined values.

VaginAY, 2017-11-16:
?I think I'm ready to try it on scala. What do you think about it?

ccelio, 2017-11-16:
What you are talking about is implementing a `valid` bit for every BTB entry, that is initialized to `false`, but once the entry is written to, the valid bit is from then on `true` for the remainder of the execution. That should deal with the 4-state `X`'s, but is very expensive in my opinion. It also adds logic to a very critical path. I'll ask around and see what advice other people have on this issue.

ccelio, 2017-11-17:
Okay, I'm going to close this issue. X-pessimism is a flaw in Verilog semantics and we get around this by using randomized initial values for things like SRAMs.  For Chisel-generated designs, randomized initial values should always be used, even when using 4-state simulation.  Initializing the branch prediction tables, even with just using a `valid` bit per entry, is too expensive.

closed 2017-11-17 by ccelio