riscv-boom/riscv-boom #44: 0x7ffff9a0 + 0x724 = 0xffffffffffe000c4 instead 0x800000c4

pentin-as, 2017-11-15:
boom-template 489f49ff2380806a8aa186416cbdb046adfee619
boom: template 1eb08e0d4c98607997874639ba9b94c44f010884
rocket-chip: master 7e75d63ba6b4c1b50aaaf920e1c693ef6acf51d7

    800029a0:	ffffd797          	auipc	a5,0xffffd
    800029a4:	72478793        addi	a5,a5,1828 # 800000c4 <trap_entry>

pentin-as, 2017-11-15:
alu in csr_exe_unit keeps the signal valid two cycles. As a result, regfile has two entries and the a5 register references an invalid entry.

[image]

ccelio, 2017-11-15:
Can you break this down into a small, reproducible test case? 

Can you show the iregfile writeports in the waveform that show the write enable and write address?

ccelio, 2017-11-15:
Sorry, I'm confused by what the issue is. Is the problem that an addition of two numbers computes the wrong answer, or is the problem the ALU asserts valid for two cycles when it should only assert valid for one cycle?

pentin-as, 2017-11-15:
regfile
[image]

information is set in two registers: 6 and 14

pentin-as, 2017-11-15:
The waveform resp_valid is equal to the waveform of the req_valid signal.

[image]

I will continue to analyze.

pentin-as, 2017-11-15:
Sorry. I think I was wrong. Temporarily close the issue.

closed 2017-11-15 by pentin-as

pentin-as, 2017-11-16:
Yes, I was wrong. Sorry.