riscv-boom/riscv-boom #47: Eternal ebreak

pentin-as, 2017-11-23:
I'm testing Boom with JTAG on FPGA. I'm set breakpoint by means of GDB (GDB replaces the instruction on which breakpoint stands for the ebreak instruction, and the replaced instruction executes in debug mode from address 0x800). However, if I call the run command, the breakpoint is called again (the execution should continue with the next instruction). Thus, execution can not be continued until there is an ebreak instruction. If I remove the breakpoint, the execution continues.

I need time to model this problem. However, there is a problem.

ccelio, 2017-11-23:
What commit hash of BOOM and rocket-chip/boom-template are you using? I just fixed an ecall bug a few hours ago (5af8a18350b55d36fdb67bb95878d0e6ce83ab93) and an ebreak fix a few weeks ago, if you are using the priv 1.10 spec.

pentin-as, 2017-11-23:
I took a boom from the devel branch.
71146913dc041362490893b413bae38dc10fd5f7

ccelio, 2017-11-25:
I see, the `devel` branch is a little stale right now.  Since it looks like your version of boom is implementing the priv 1.10 spec (and the latest rocket-chip changes), then you'll want to update to the latest commit that's on the `template` branch (https://github.com/ucb-bar/riscv-boom/tree/template), which includes some Priv 1.10 fixes to illegal instruction handling and ecall handling. 

However, the `template` branch is set up to work with a different base repository: (https://github.com/ccelio/boom-template) instead of rocket-chip.

I have one remaining performance change to make to the `template` branch and then I'll push it to stable master (and update devel to match). I'm guessing 1-2 weeks.

Let me know if this fixes your issues.

pentin-as, 2017-12-07:
Sorry for so long.

I confirm that in:
boom-template master f50c1a6
boom template 52b2bda
rocket-chip master 5155eb6
... there is this problem.

abejgonzalez, 2019-05-28:
@pentin-as Is this solved already? If so I will close this issue.

pentin-as, 2019-05-29:
I will check next week. If left, I'll fix it.

closed 2021-04-05 by jerryz123