riscv-boom/riscv-boom #48: [btb-sa] Missing io.req.bits.addr port and last_pc loop without init

VaginAY, 2017-11-26:
This problem was discovered after the synthesis in gatelevel netlist and run with
memory randomization without registers randomization.
/* For this, I modified the factory memory models, but I did not touch the model of standard cells.*/
After a short investigation, the problem was confirmed in the RTL code with the randomization of registers disabled.

Here is sch of p_table read address driver:
[image]

And here is sch of io_s1_r_idx[14:0]  from previous picture driver:
[image]

btb-sa.scala code:
```
   val s0_pc = Wire(UInt(width=vaddrBits))
   val last_pc = RegNext(s0_pc)
   s0_pc := Mux(stall, last_pc, io.req.bits.addr)
   val s1_pc = RegNext(s0_pc)
   s1_resp_bits.fetch_pc := s1_pc
   io.s1_pc := s1_pc
```

As you can see, there is no any mux on s0_pc reg.
When I checked the generated code of BTBsa.v, I saw that there was no io.req.bits.addr port.

When I checked scala source of this signal, I saw
`btb.io.req := io.ext_btb_req` in bpd-pipeline.scala
But there is no any assignments to ext_btb_req.

As I understand it, the problem is a broken predictor, which was discussed in #38.
When do you plan to correct the predictor? I doubt that I can help you, but if I can be of any use, then I am ready to try.

Best regards, Vagin Alexander.

ccelio, 2017-11-26:
You are correct, the BTB in `template` branch is disabled and the `ext_btb_req` signals are currently dangling. Fixing this is the current item on my todo list.

ccelio, 2018-03-12:
This issue should no longer be relevant. The BTB is now re-enabled.

closed 2018-03-12 by ccelio