riscv-boom/riscv-boom #49: [synth] Gatelevel netlist frequency below expected

VaginAY, 2017-11-26:
I tried to perform the synthesis of ExampleBoomSystem with 64Kx64 SimAXIMem in the technological basis of 90 nm.
Here is results (worst case):
Area - ~40 mm2;
Number of instances - 796290;
Frequency - not more than 140 MHz.

Expected Frequency was 200-400 MHz.
Сritical paths include different combinatorial paths through several stages of the pipeline.
For example fetch->decode->lsu or fetch->decode->rename.
Maybe others, I analyzed only the first 100.

Have you perform the synthesis before? If so, what results did you get? Is the current version of the code optimal, in your opinion? Or is there an optimization capability?

ccelio, 2017-11-26:
BOOM has been taped-out in TSMC 28 nm HPM. The post-synthesis results were ~37 fan-out-of-4 and the post-place-and-route results were ~50 fan-out-of-4. This was performed using the MediumBoomConfig. 

There are a number of modules that your tools need to retime. This is what we retimed, but you may find value in retiming more:

````
RenameStage
IntToFP*
 FPToFP*
FPUFMAPipe*
IMul*
BranchPredictionStage
ALUUnit
````

Is your area including 4 MB? That sim memory is just a simulation construct. And you will probably find that you cannot synthesis a 9-port register file without hitting some serious place-and-route issues. You will probably want to blackbox that and provide your own implementation (as well as provide your own SRAMs for many of the blackboxed SeqMems).

The current expected critical paths are from the load writeback path (D$->shim->register file), although there are others that are on my radar.

Some information can be found here:
https://carrv.github.io/papers/celio-boom2-carrv2017.pdf

VaginAY, 2017-11-27:
1. FO4 for 28 nm is about 10 picoseconds. This means that the frequency you reached was about 2 GHz?
2. FO4 for 90 nm is about 30 picoseconds. This means that the maximum frequency of our project will not exceed 650 MHz. This is much more, then we expect.
3. Thank you for list of modules for retiming. But you did not exactly carry out this procedure for Issue modules?
4. Yes, of course, all SeqMems (including SimAXIMem mem_ext block) was blackboxed and replaced by fab RAM compiler's blocks. 64Kx64(4 Mbit) included in 40 mm2.

ccelio, 2017-11-27:
> 1. This means that the frequency you reached was about 2 GHz?

We used 28 nm HPM, which I don't think is quite as fast as you are quoting.  I'll provide more specific numbers at a later date when I'm allowed to.

> 2. This is much more, then we expect.

We'll see, physical design issues can be troublesome. Keep me posted though.

> 3. But you did not exactly carry out this procedure for Issue modules?

I didn't feel a need to retime the issue queues after giving them a full clock cycle to perform issue-select. You can of course feel free to retime many more modules than I did.

> 4. 64Kx64(4 Mbit) included in 40 mm2.

Hmmm, the chip-top should be basically just an AXI port and a JTAG port, and the SimAXIMem should be off-chip/external. But do whatever works for you.

VaginAY, 2017-11-29:
The results after the retiming are much better. However, there are paths that the length exceeds 100 FO4. I added the list of the first 100 to the attachment.
[crit_paths.txt](https://github.com/ucb-bar/riscv-boom/files/1512789/crit_paths.txt)

ccelio, 2017-11-29:
This is DefaultBoomConfig? Can you try MediumBoomConfig and see if you notice a difference?

And can you give me the full path from:

    Startpoint:  ExampleBoomSystem/tile/boom/core/iregister_read/exe_reg_uops_1_fu_code_reg_9/CP
    Endpoint:  ExampleBoomSystem/tile/boom/core/issue_units_1/IssueSlot_13/slot_p1_reg/D

I've not seen that path before.

VaginAY, 2017-11-30:
The results of the proceedings showed that the problem was in the set of libraries. Sorry for the disinformation about first case.
But second case still actual.

ccelio, 2017-11-30:
Okay thanks for the data, that looks a lot more like what I expect.

First, you will almost certainly need to blackbox the integer register file. This is described a bit in the above CARRV paper. You can see some of our Chisel models of these block boxes here (https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/regfile-custom.scala), but you will need to provide your own Verilog code/macro.

Second, the TLBuffer/mshr/shim/mem_unit/iregfile is the known, current problematic path. That will get better when you do more physical design on your register file, but it will be a while before I provide a better implementation of that path.

Also, have you tried rocket using DefaultConfig, instead of boom? I'm interested in your experience using a known-good design that's been taped out in multiple processes.

VaginAY, 2018-02-18:
Good day, some new information after delay.

1. Max freq on 90 LP tech is 180 MHz. 1FO4 is 48.5 ps.
So ~124FO4 for design.
Rocket give us about ~ 60FO4 (~300 Mhz).
2. Topology prototyping boom with rocket environment system give us 9x9mm chip without
any problems in tracing (seven metals). Memory included.
3. Problem critical paths: 
  3.1 LSU.
  3.2 Issue unit
  3.3 D$->shim->register file and similar
// Another problem is memory, but it is not boom problem

But if 3.1-3.3 with memory problems will be solved, we see, that we can achieve 300-500 MHz!
That's cool!

For now Pentin Alexander (pentin-as) is working on optimization.

ccelio, 2018-02-18:
What configuration are you using? Are you using DefaultBoomConfig? For 3.2, I would expect that you probably need to shrink the size of the Issue Units which should help the critical path there. Likewise, 3.1 may also require a smaller LSU (num_lsu_entries I think).  3.3 is a known problem. I'm not yet sure if I can fix it as is, or if I need to add another stage for the sign-extension/formatting of load data. Unfortunately it is a long path that interacts with the physical design of the register file.

If you can provide critical path reports that would be great, but I probably won't be able to get to addressing them for another month or two. Keep me posted on what changes you make if you end up addressing these issues.