riscv-boom/riscv-boom #51: make run CONFIG=BOOMConfig error

shruthi1996, 2017-12-06:
Hi, I tried installing BOOM for my assignment and I have an error while running the make command (make run CONFIG=BOOMConfig).
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-BOOMConfig' failed
make: *** [emulator-rocketchip-BOOMConfig] Error 137

I tried the rocket-chip installation previously which worked fine. I used the git links https://github.com/freechipsproject/rocket-chip and https://github.com/riscv/riscv-tools.

Below is the complete part of the error when compiled the command:

mkdir -p verilator/
wget http://www.veripool.org/ftp/verilator-3.884.tgz -O verilator/verilator-3.884.tar.gz
--2017-12-05 12:20:55-- http://www.veripool.org/ftp/verilator-3.884.tgz
Resolving www.veripool.org (www.veripool.org)... 173.230.151.124
Connecting to www.veripool.org (www.veripool.org)|173.230.151.124|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.veripool.org/ftp/verilator-3.884.tgz [following]
--2017-12-05 12:20:55-- https://www.veripool.org/ftp/verilator-3.884.tgz
Connecting to www.veripool.org (www.veripool.org)|173.230.151.124|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2027268 (1.9M) [application/x-gzip]
Saving to: ‘verilator/verilator-3.884.tar.gz’

verilator/verilator-3.884.tar.gz 100%[================================================================================================================>] 1.93M 1.93MB/s in 1.0s

2017-12-05 12:20:57 (1.93 MB/s) - ‘verilator/verilator-3.884.tar.gz’ saved [2027268/2027268]

rm -rf verilator/src/verilator-3.884/
mkdir -p verilator/src/verilator-3.884/
cat verilator/verilator-3.884.tar.gz | tar -xz --strip-components=1 -C verilator/src/verilator-3.884/
touch verilator/src/verilator-3.884/configure
mkdir -p verilator/src/verilator-3.884/
cd verilator/src/verilator-3.884/ && ./configure --prefix=/home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install
configuring for Verilator 3.884 2016-05-18
checking whether to disable hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking that C++ compiler can compile simple program... yes
checking for perl... /usr/bin/perl
checking for flex... /usr/bin/flex
checking for bison... /usr/bin/bison
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for size_t... yes
checking for size_t... (cached) yes
checking for inline... inline
checking whether g++ accepts -Wno-unused-parameter... yes
checking whether g++ accepts -Wno-char-subscripts... yes
[... 294 lines cut ...]
; for p in include/.[chv] include/vltstd/.[chv] ; do
/usr/bin/install -c -m 644 $p /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/verilator/$p;
done
/bin/sh ./mkinstalldirs /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/verilator/examples/test_c
/bin/sh ./mkinstalldirs /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/verilator/examples/test_sc
/bin/sh ./mkinstalldirs /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/verilator/examples/test_v
cd .
; for p in test_v/.[chv] test_c/.[chv] test_c/Makefile test_c/Makefile_obj test_sc/.[chv] test_sc/Makefile test_sc/Makefile_obj ; do
/usr/bin/install -c -m 644 $p /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/verilator/examples/$p;
done
/bin/sh ./mkinstalldirs /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/pkgconfig
/usr/bin/install -c -m 644 verilator.pc /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/share/pkgconfig
make[1]: Leaving directory '/home/shruthi/Documents/ca/rocket-chip/emulator/verilator/src/verilator-3.884'
touch /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/bin/verilator
mkdir -p /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/shruthi/Documents/ca/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.fir -o /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v -X verilog
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
mkdir -p /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig
/home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=$c("verbose","&&","done_reset") +define+STOP_COND=$c("done_reset") --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/shruthi/Documents/ca/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/shruthi/Documents/ca/rocket-chip/riscv-tools//include -I/home/shruthi/Documents/ca/rocket-chip/csrc -DVERILATOR -include /home/shruthi/Documents/ca/rocket-chip/csrc/verilator.h" -Mdir /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig
-o /home/shruthi/Documents/ca/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/shruthi/Documents/ca/rocket-chip/csrc/emulator.cc /home/shruthi/Documents/ca/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -Wl,-rpath,/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -L/home/shruthi/Documents/ca/rocket-chip/emulator -lfesvr -lpthread"
-CFLAGS "-I/home/shruthi/Documents/ca/rocket-chip/emulator/generated-src -include /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h"
Killed
%Error: Command Failed /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/shruthi/Documents/ca/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/shruthi/Documents/ca/rocket-chip/riscv-tools//include -I/home/shruthi/Documents/ca/rocket-chip/csrc -DVERILATOR -include /home/shruthi/Documents/ca/rocket-chip/csrc/verilator.h' -Mdir /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig -o /home/shruthi/Documents/ca/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/shruthi/Documents/ca/rocket-chip/csrc/emulator.cc /home/shruthi/Documents/ca/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -Wl,-rpath,/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -L/home/shruthi/Documents/ca/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/shruthi/Documents/ca/rocket-chip/emulator/generated-src -include /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h'
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-BOOMConfig' failed
make: *** [emulator-rocketchip-BOOMConfig] Error 137

ccelio, 2017-12-06:
Can you try typing "make" again and see if the error message is shorter? I'm having trouble seeing anything that is obviously wrong.

shruthi1996, 2017-12-06:
Yup, I tried make and this is what I got
mkdir -p /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig
/home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/shruthi/Documents/ca/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/shruthi/Documents/ca/rocket-chip/riscv-tools//include -I/home/shruthi/Documents/ca/rocket-chip/csrc -DVERILATOR -include /home/shruthi/Documents/ca/rocket-chip/csrc/verilator.h" -Mdir /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig \
-o /home/shruthi/Documents/ca/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/shruthi/Documents/ca/rocket-chip/csrc/emulator.cc /home/shruthi/Documents/ca/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -Wl,-rpath,/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -L/home/shruthi/Documents/ca/rocket-chip/emulator -lfesvr -lpthread" \
-CFLAGS "-I/home/shruthi/Documents/ca/rocket-chip/emulator/generated-src -include /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h"
Killed
%Error: Command Failed /home/shruthi/Documents/ca/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/shruthi/Documents/ca/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/shruthi/Documents/ca/rocket-chip/riscv-tools//include -I/home/shruthi/Documents/ca/rocket-chip/csrc -DVERILATOR -include /home/shruthi/Documents/ca/rocket-chip/csrc/verilator.h' -Mdir /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig -o /home/shruthi/Documents/ca/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/shruthi/Documents/ca/rocket-chip/csrc/emulator.cc /home/shruthi/Documents/ca/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -Wl,-rpath,/home/shruthi/Documents/ca/rocket-chip/riscv-tools//lib -L/home/shruthi/Documents/ca/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/shruthi/Documents/ca/rocket-chip/emulator/generated-src -include /home/shruthi/Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h'
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-BOOMConfig' failed
make: *** [emulator-rocketchip-BOOMConfig] Error 137

pbazarnik, 2017-12-06:
For what it's worth: 
I have seen similar failure caused by Makefrag-verilator trying to download specific Verilator version and failing (see below)
I have found this version of Verilator in Ubuntu src repo and replaced wget by copy:

verilator/verilator-$(VERILATOR_VERSION).tar.gz:
	mkdir -p $(dir $@)
	cp ~/Downloads/verilator_3.884.orig.tar.gz $@
#	wget http://www.veripool.org/ftp/verilator-$(VERILATOR_VERSION).tgz -O $@ <-- this fails to download

ccelio, 2017-12-06:
Interesting, all rocket-chip stuff that I can find use Verilator 3.904 (not that I think it should matter). The error message definitely shows that... something... has gone wrong with verilator, but it doesn't give me enough information to know what. :(

pbazarnik, 2017-12-06:
That was some time ago and the Verilator version has likely changed since than (I used master)

I just searched the log and it looks like Verilator is building: (3.884) so wget should be ok.

`make[3]: Entering directory '/home/shruthi/Documents/ca/rocket-chip/emulator/verilator/src/verilator-3.884/src/obj_dbg' `

shruthi1996, 2017-12-06:
Yes, the problem is with verilator because when i tried running the cycle accurate verilator in rocket-chip, I got an error in make as well:

make: *** No rule to make target '/path/to/install/riscv/toolchain/riscv64-unknown-elf/share/riscv-tests/isa/rv64uf-p-ldst', needed by 'output/rv64uf-p-ldst'.  Stop.

[image]

pbazarnik, 2017-12-07:
This message refers to building a test which would use  gcc toolchainno Verilator
> No rule to make target '/path/to/install/riscv/toolchain/riscv64-unknown-elf/share/riscv-tests/isa/rv64uf-p-ldst

In your initial log there is a message that verilator_bin process was "Killed"
> ...../Documents/ca/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h"
> Killed
> 
This possibly could happen if you run out of RAM and have no swap space (BOOM compilation may take few GB or RAM)
You can try building the SmallBOOM (it takes less RAM)

make run CONFIG=SmallBoomConfig

spandana05, 2017-12-07:
Hi,
Even I am getting the same error
mkdir -p /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig
/home/spandana/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/spandana/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/spandana/rocket-chip/riscv/include -I/home/spandana/rocket-chip/csrc -DVERILATOR -include /home/spandana/rocket-chip/csrc/verilator.h" -Mdir /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig \
-o /home/spandana/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/spandana/rocket-chip/csrc/emulator.cc /home/spandana/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/spandana/rocket-chip/riscv/lib -Wl,-rpath,/home/spandana/rocket-chip/riscv/lib -L/home/spandana/rocket-chip/emulator -lfesvr -lpthread" \
-CFLAGS "-I/home/spandana/rocket-chip/emulator/generated-src -include /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h"
Killed
%Error: Command Failed /home/spandana/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/spandana/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/spandana/rocket-chip/riscv/include -I/home/spandana/rocket-chip/csrc -DVERILATOR -include /home/spandana/rocket-chip/csrc/verilator.h' -Mdir /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig -o /home/spandana/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/spandana/rocket-chip/csrc/emulator.cc /home/spandana/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/spandana/rocket-chip/riscv/lib -Wl,-rpath,/home/spandana/rocket-chip/riscv/lib -L/home/spandana/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/spandana/rocket-chip/emulator/generated-src -include /home/spandana/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h'
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-BOOMConfig' failed
make: *** [emulator-rocketchip-BOOMConfig] Error 137
I tried using     make run CONFIG=SmallBoomConfig
I am getting an error as:
[info] Running rocketchip.Generator /home/spandana/rocket-chip/emulator/generated-src rocketchip TestHarness boom SmallBOOMConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "boom.SmallBOOMConfig" from "SmallBOOMConfig", did you misspell it?
java.lang.Exception: Unable to find part "boom.SmallBOOMConfig" from "SmallBOOMConfig", did you misspell it?
	at Chisel.package$throwException$.apply(compatibility.scala:208)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:32)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at rocketchip.Generator$.getConfig(Generator.scala:11)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at rocketchip.Generator$.config$lzycompute(Generator.scala:11)
	at rocketchip.Generator$.config(Generator.scala:11)
	at util.GeneratorApp$class.world(GeneratorUtils.scala:80)
	at rocketchip.Generator$.world$lzycompute(Generator.scala:11)
	at rocketchip.Generator$.world(Generator.scala:11)
Caused by: java.lang.ClassNotFoundException: boom.SmallBOOMConfig
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:29)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at rocketchip.Generator$.getConfig(Generator.scala:11)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at rocketchip.Generator$.config$lzycompute(Generator.scala:11)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
[... cut ...]
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 10 s, completed Dec 7, 2017 5:25:18 AM
Makefrag-verilator:12: recipe for target '/home/spandana/rocket-chip/emulator/generated-src/rocketchip.SmallBOOMConfig.fir' failed
make: *** [/home/spandana/rocket-chip/emulator/generated-src/rocketchip.SmallBOOMConfig.fir] Error 1

Thanks in advance

pbazarnik, 2017-12-07:
Based on log you make SmallBOOMConfig instead of SmallBoomConfig

SmallBOOMConfig is correct (case sensitive)

> I tried using make run CONFIG=SmallBoomConfig
> I am getting an error as:
> [info] Running rocketchip.Generator /home/spandana/rocket-chip/emulator/generated-src rocketchip TestHarness boom [...]

spandana05, 2017-12-07:
I am still getting the same error even though I used        make run CONFIG=SmallBOOMConfig

VaginAY, 2017-12-08:
Good day. You should use 
`make run CONFIG=SmallBoomConfig`
no `make run CONFIG=SmallBOOMConfig`
I just checked, all works fine.

spandana05, 2017-12-09:
It worked.. Thanks a lot

shruthi1996, 2017-12-09:
yes it worked, thank you so much

ccelio, 2018-03-12:
Thanks everybody. Glad we were able to sort out the verilator issues.

closed 2018-03-12 by ccelio