riscv-boom/riscv-boom #52: LSU Questions

pentin-as, 2017-12-15:
I. Duration of the reset signal.
I have a test environment, which before each test sets a reset signal and zeroes the bss section. If the reset signal lasts one or two cycles, exception occurs:
Assertion failed: stq_execute_head got off track.
    at slu.scala:498 assert (stq_entry_val(stq_execute_head) ||
sometimes:
Assertion failed: [rob] writeback (1) occurred to an invalid ROB entry.
    at rob.scala:579 assert (!(io.wb_resps(i).valid && MatchBank(GetBankIdx(rob_idx)) &&
If the reset signal lasts 3 cycles, all tests pass.
In https://github.com/freechipsproject/rocket-chip/issues/1152 I was told that the reset signal should be 16 cycles. Does this policy apply to Boom too?

II. TLB in LSU.
The LSU has a TLB. Its outputs are connected to io.memreq_addr:
exe_tlb_paddr = Cat(dtlb.io.resp.paddr(paddrBits-1,corePgIdxBits), exe_vaddr(corePgIdxBits-1,0))
val exe_ld_addr = Mux(will_fire_load_incoming || will_fire_load_retry, exe_tlb_paddr, laq_addr(exe_ld_idx_wakeup))
io.memreq_addr    := exe_ld_addr
...and leave the core...
exe_units.memory_unit.io.lsu_io := lsu.io
io.dmem.req.bits.addr  := io.lsu_io.memreq_addr
dc_shim.io.core <> exe_units.memory_unit.io.dmem
io.dmem <> dc_shim.io.dmem
...and get to the input vaddr DCache, and then go to another TLB in DCache...
dcachePorts += core.io.dmem
dcacheArb.io.requestor <> dcachePorts
outer.dcache.module.io.cpu <> dcacheArb.io.mem
when (io.cpu.req.valid) {
  s1_req := io.cpu.req.bits
}
dtlb.io.req.bits.sfence.bits.addr := s1_req.addr
s1_addr = dtlb.io.resp.paddr
...in which virtual addresses are again converted into physical addresses.
In addition, in Boom PTW has three inputs: from TLB in LSU, from TLB in DCache and from TLB in ICache. In the Rocket in PTW, there are only two inputs: from TLB in DCache and from TLB in ICache.
I think there is a mistake.

III. Why are LSU queues initialized at the decoder stage?
It seems to me it is the following problems:
1. Inefficient use of queues. If instructions enter into the decoder that do not need to be executed (due to a miss of the branch predictors), the slot in the queue becomes invalid.
   // Kill speculated entries on branch mispredict
   for (i <- 0 until num_ld_entries)
   {
      when(laq_allocated(i))
      {
         laq_uop(i).br_mask := GetNewBrMask(io.brinfo, laq_uop(i))
         when (IsKilledByBranch(io.brinfo, laq_uop(i)))
         {
            laq_allocated(i)   := Bool(false)
            laq_addr_val(i)    := Bool(false)
         }
      }
   }
But the head of the queue does not shift. This problem would not exist if the queue was initialized at a later stage of the pipeline (when computing the address).
2. A large number of combinatorial logic. It seems to me that for MegaBoom, this code:
for (w <- 0 until pl_width)
   {
      ...
      ld_enq_idx = Mux(io.dec_ld_vals(w), WrapInc(ld_enq_idx, num_ld_entries),
                                          ld_enq_idx)
      ...
      st_enq_idx = Mux(io.dec_st_vals(w), WrapInc(st_enq_idx, num_st_entries),
                                          st_enq_idx)
  }
... will create a difficult critical path. With that, the addresses from MemUnit all the same arrive consistently.
3. Increase the number of registers, because the decoder contains information about where the information entered in the LSU queue (ldq_idx). And this information should be transferred to other stages of the pipeline.
But I do not see any benefit from initialization at such an early stage.

IV. Why does configs specify a size issue for MEM?
IssueParams(issueWidth=1, numEntries=20, iqType=IQT_MEM.litValue)
This parameter is not used, because this issue does not exist, because the queue for the memory operation is in the LSU (and also configured in configs).

ccelio, 2017-12-16:
**1. reset signal**

I believe the test harness holds reset for 10 cycles, but there isn't some magic number the design is aiming for. If you see some error that's causing an assertion to trigger, it could be a mistake that we should fix.

**TLB**

The TLB within the D$ is hard-coded to never perform translations, so it should not be lengthening the critical path to perform an additional translation. 

I should explicitly address the PTW port count; the unused TLB should never emit a request, but I could be making a mistake here since I haven't verified that the tools prune it away.

**LSU**

This code shifts the tail pointer on a branch mispredict to reclaim the mispredicted slots (https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/lsu.scala#L1053-L1057)

IQT_MEM holds the memory instructions whose registers are not yet ready. The LSU holds the computed addresses of the memory instructions.

poweihuang17, 2017-12-28:
Really cool!

pentin-as, 2018-02-07:
1. Fixed in https://github.com/ucb-bar/riscv-boom/pull/59.
4. I was wrong about Issue MEM. Issue is. I do not know why I did not notice.

closed 2018-10-26 by jerryz123