riscv-boom/riscv-boom #55: NLP

pentin-as, 2018-01-11:
1. Can I rename BTBsa to NLP in all source code in accordance with the BOOM Design Specification?

2. I see that because NLP is out of Frontend, it works slower than BTB in Rocket, which is in Frontend (I can attach pictures to make it clear).  In addition, the BOOM Design Specification says that NLP must be in Frontend.
I understand correctly that the work on the transfer of NLP in Frontend is planned, but not yet conducted?
This is what you meant when you wrote that for Boom will be expanding the interface into Frontend to add additional signals (https://github.com/ucb-bar/riscv-boom/pull/53)?

3. NLP Boom is different from BTB Rocket. This is because NLP was copied earlier and obsolete, so it should be updated (with large buffer sizes in parameters)? Or is the NLP specifically made different from the BTB Rocket?

ccelio, 2018-01-11:
I'm sorry that the BOOM Design Specification (BDS) has become out-of-date. To help, I recommend using (https://www2.eecs.berkeley.edu/Pubs/TechRpts/2017/EECS-2017-157.html) as an additional reference that supersedes the BDS. Unfortunately, I suspect I won't get to updating the BDS for a few more months.

I have decided on a new terminology that I believe is better reflected by what other companies use to describe these common structures.

a micro-BTB (uBTB, µBTB): a tagged target predictor that operates in a single cycle, making predictions with no bubbles. 

a BTB: a tagged target predictor that operates in two cycles (or more cycles), making a prediction with one (or more) bubbles. 

Some designs may choose to have both a µBTB and a main BTB, especially if they want to look good on some particular micro-benchmark. 

BOOM's BTB differs from Rocket's in many ways: it uses SRAM (not flip-flops), it is set-associative (not FA), and it uses partial tags (not full tags). It is designed to store an order of magnitude more entries for a given clock frequency. To that end, BOOM's BTB also uses a less sophisticated bimodal predictor instead of gshare (and simplifies by removing the need to track/maintain global history).

For BOOMv2, I had to use a forked rocket-chip Frontend. With the new BOOMv3/boom-template work, I'm trying to use rocket-chip master branch, which means I must copy the rocket-chip Frontend into riscv-boom so I can make the necessary changes there (i.e., I need access to pc_s0), among other changes to improve superscalar fetch.

I am also currently updating BOOM's BTBsa to make predictions in two cycles (one bubble on a prediction), which is a better improvement to the timings of accessing SRAMs and selecting the next PC.

I named BOOM's BTB as `BTBsa` in part to make sure I did not have any accidental name space collisions with Rocket. I may eventually rename it simply `BTB` at some later date.

I hope this helps clarify my thinking and direction.

pentin-as, 2018-01-16:
Thank you very much. I would like to ask two more questions:

Question 1.
Here is the source code:
0000000080000000 <_start>:
    80000000:	00000013          	nop
    80000004:	ffdff0ef          	jal	ra,80000000 <_start>
    80000008:	00000013          	nop
    8000000c:	00000013          	nop
    80000010:	00000013          	nop
    80000014:	00000013          	nop
    80000018:	00000013          	nop
    8000001c:	00000013          	nop
    80000020:	00000013          	nop
    80000024:	00000013          	nop
    80000028:	00000013          	nop
    8000002c:	00000013          	nop
    80000030:	00000013          	nop
    80000034:	00000013          	nop
    80000038:	00000013          	nop
    8000003c:	00000013          	nop

This is Frontend Rocket:
[image]

This is Frontend MegaBoom:
[image]

From the frontend Rocket each cycle the instruction outputs (with addresses 8000_0000 and 8000_0004). Predictions are carried out inside Frontend.

In the case of Boom:
s0 - frontend outputs.
s1 - BTBsa in BranchPredictorStage makes a prediction and, through FetchUnit, sends a new PC value to Frontend.
s2 - the bubble needed for Frontend to send a new request to ICache.

Thus, predictors in Rocket are faster than in Boom.

I believe that this can be corrected if either:
 - move BTBsa to Frontend (not our option);
 - eliminate Frontend completely, i.e. leave only ICache + TLB in it, and completely transfer the PC to FetchUnit.

What do you think about it?

Question 2.
Could you recommend me a good book (in English) about BTB, GSkew, GShare, TAGE and other predictors.

pentin-as, 2018-01-16:
Please excuse me. I was a bit confused in terminology and did not understand that it was BTB (I thought that GShare) was disabled in commit https://github.com/ucb-bar/riscv-boom/commit/ebbfee5208efdb143bf85352c2bb05105c9b17e4. And according to https://github.com/ucb-bar/riscv-boom/issues/48 disabled BTB does not generate s1_pc and GShare does not work too. In this case, uDec predicts. However, still they all use FetchUnit and deal with Frontend. So the question remains.

ccelio, 2018-01-16:
I am currently working to re-enable the BTB in the boom `template` branch. In Rocket, the BTB is implemented using combinational logic (and flip-flops), so it is able to redirect the icache very fast, and leaves no bubbles in the pipeline (a "micro-BTB"). But it must be very small to do so. For BOOM, its BTB is implemented using SRAM, so it will redirect the icache and leave one bubble. 

I don't know of a good book that describes branch prediction. Instead, I would look at lecture slides and the original research papers.

http://www-inst.eecs.berkeley.edu/~cs152/sp18/
http://csg.csail.mit.edu/6.823/lecnotes.html

sorear, 2018-01-16:
As long as the next PC can be computed combinationally from the current PC and the output of the BTB SRAM it should still be possible to do this without bubbles no?

ccelio, 2018-01-16:
// 3 stage pipeline
    S0: next-pc
    S1: I$ ram access / BTB ram access
    S2: redirect pc on BTB prediction

The problem is there is not enough time to take the output from the BTB SRAM (which requires tag hit, way-selection, and muxing) and then go through the next-pc mux logic. Therefore the btb redirect is moved back to the S2 stage.

The recent BOOM tapeout does indeed redirect the frontend using BTB SRAM with no bubbles, but it was a critical path and it limits how fast the processor can go, so I'm changing that logic going forward. Sure, what works best is a matter of aggressiveness of the design, but I will content that little noticeable performance will be lost by adding a bubble (and one could always add a µBTB/next-line-predictor in concert with the main BTB if it really was an issue).

sorear, 2018-01-16:
Makes sense, I always forget way selection is costly

pentin-as, 2018-01-17:
But now uDec predicts. It works on combinatorial logic. But there is a bubble. I have fears that when BTB is repaired there will be two bubbles.

ccelio, 2018-03-12:
I have now pushed to master an update with a working BTB (1 bubble is inserted on a taken prediction). The work is still Work-in-Progress, but I will close this issue for now.

closed 2018-03-12 by ccelio