riscv-boom/riscv-boom #56: Pipeline Hang

amerlo94, 2018-01-16:
Hi,

I would like to report an issue that I am facing with the latest commit of the boom-devel branch in the rocket-chip generator repository (i hope this is the right place to ask).

I clone the rocket-chip repository and checkout boom-devel branch. I need the boom-devel one since I saw that the boom branch is not update to the latest configuration changes for the rocket core and I need consistency between the two cores.

I wanted to test it out with the provided emulator, so I issue the following make commands in the emulator folder:
make output/dhrystone.riscv.out

It should emulate a dhrystone code execution on the default configuration: BoomConfig.

But it ends up in the following error (in the dhrystone.riscv.out file generated):
using random seed 1516081992
Assertion failed: Pipeline has hung.
    at core.scala:1033 assert (!(idle_cycles.value(13)), "Pipeline has hung.")

Is this something related to the boom-devel branch, or I am missing something?

Working with the following riscv-boom repository commit:
98bae1f6b066d491c6e07e181e9bef3e73587ff6

And the following rocket-chip repository commit:
f03ab95deed4e41bffa252de34fd9fdb93cda6e5

Thanks

pentin-as, 2018-01-16:
Try to take the boom and rocket from the boom-template repository:

git clone https://github.com/ccelio/boom-template.git
cd boom-template
./scripts/init-submodules.sh

amerlo94, 2018-01-16:
Hy,

thanks for the suggestion.

I tried with this other repository and I got that:
it works with a simple isa test as rv64ui-p-simple, but it fails (with same error) with the dhrystone benchmarks.

amerlo94, 2018-01-16:
In addition I generated verilog src files for the Boom core and simulated in Modelsim. I have a working Rocket core simulated in this way and everything is working fine.

But as soon as I instantiated the Boom core and halt hart 1 via the DMI interface, I see lot of "X" in the internal core signals and than the pipeling hung error. Is there any debug procedure in order to interact with the Boom core?

I thought it was the same as the Rocket core...

pentin-as, 2018-01-17:
1. I have Dhrystone working on a verilator with MegaBoomConfig. Try:
make CONFIG=MegaBoomConfig
make run-bmark-tests CONFIG=MegaBoomConfig

2. DMI is completely borrowed from Rocket. At me in CAD Cadence DMI works.
Is randomization enabled (on the Rocket without it could not work)?
I ran with randomization disabled. But for this in the test environment I had to write:
testchip_tb.dut.ExampleBoomSystem.tile.boom.core.bpd_stage.btb.last_pc = 39'h0;
testchip_tb.dut.ExampleBoomSystem.tile.boom.ptw.arb.lastGrant = 2'h0;

3. At present, the results of Dhrystone for MegaBoomConfig are 1.38 times worse than those of Rocket. Do not be surprised. I think this is due to the fact that currently predictors are temporarily disabled in Boom.

amerlo94, 2018-01-17:
1. I am on the last commit of boom-template, and still it hung on the first benchmark test, the mm.riscv one.

2. As for the DMI side, I noticed that the core has always the "allhavereset" bit set to 1 in the "Debug Module Status" Debug Module register. Is it ok?

donggyukim, 2018-01-17:
Make sure RVC is off in gcc:
https://github.com/ccelio/boom-template/blob/master/scripts/build-tools.sh

codelec, 2018-01-17:
i am able to reproduce similar results i.e. bmarks having pipeline hung. there is no difference in how bmark or an asm test is handled by debug module 

allhavereset being set indefinitely is not a good sign probably SimDTM/logic in dtm.cc faced an error while loading the elf's

amerlo94, 2018-01-18:
Today I will try with RVC off in GCC.

I noticed that "allhavereset" is set from the beginning of the emulation,
thus from the first read of the "Debug Module Status" register. So i do not
think that it is a problem of SimDTM...

amerlo94, 2018-01-18:
The Pipeline Hung error still persists with RVC off in gcc.

pentin-as, 2018-01-19:
Let's do it again. Do you do something like this?

git clone https://github.com/ccelio/boom-template.git
cd boom-template
./scripts/init-submodules.sh
export RISCV=toolchain
./scripts/build-tools.sh
cd verisim
make CONFIG=MegaBoomConfig
make run-bmark-tests CONFIG=MegaBoomConfig

codelec, 2018-01-19:
@amerlo94 sorry I posted without considering donggyu's suggestion 
all asm and bmarks pass if either the compiler is built with RVC off or compiler is passed the following flag `-march=rv64imafd` . for me a better way out was to build the compiler with `-enable-multilib` in this way just by changing the `-march` it can be used with rocket, boom and also sodor

amerlo94, 2018-01-19:
Ok, I did all the flow again in a clean folder and it works.

amerlo94, 2018-01-23:
Hey,

I am experiencing a unexpected behavior of the BOOM rtl code. When I compile the emulator and test a BOOM configuration on the provided emulator, everything works. But if I use the generated verilog code from the vsim makefile, and not the verisim one (i need the other since it allows me to input technology depended memories), I came across the pipeline hung issue.

Is there any difference in the two generated verilog rtl files?

pentin-as, 2018-01-24:
As far as I understand, in terms of verilog code, the difference is only in RAM. In vsim, the RAM is in the *.behav_srams.v file (so that it can be replaced with your own). In verisim RAM is generated along with the other code.

The main difference in this line:
https://github.com/ccelio/boom-template/blob/856f043999c7ee4fea05964a276f3b1f58553fa6/vsim/Makefile#L12

I correctly heard:
 - you use your own test environment;
 - in it you use the .v file from the verisim folder and everything works;
 - when you replace it with .v and behav_srams.v files from the vsim folder, see "pipeline hangs"
?

amerlo94, 2018-01-24:
I supposed it was the only difference, but since I am experiencing this issue I was not quite sure.

Yes:

- I am using Modelsim in my own test environment
- I issue the make in the verisim folder and use the generated .v file. Boom is able to run dhrystone.
- I issue the make verilog in the vsim folder and use the generated .v file. Boom hung.

abejgonzalez, 2019-05-28:
Hi @amerlo94 is this issue now resolved? If so, I will go ahead and close this issue.

amerlo94, 2019-05-28:
Hi @abejgonzalez, yes it is.

closed 2019-05-28 by amerlo94