riscv-boom/riscv-boom #57: Is BOOM affected by Spectre and Meltdown?

sunnystormy, 2018-01-25:
Since BOOM is an Out-of-Order processor, and because Spectre and Meltdown rely upon speculative memory accesses to compromise a host system, is BOOM also affected by this? If so, are there any steps being taken to address this?

sorear, 2018-01-25:
BOOM isn't affected by Meltdown; *all* out-of-order designs are affected by some variants of Spectre attacks (there are many possible variants that are not well-known yet).  No "steps" are being taken.

tommythorn, 2018-01-25:
>
> all out-of-order designs are affected by some variants of Spectre attacks
>

I don't think that's right.  It's speculation, not out-of-order, that lead
to Spectre.
BOOM might be affected, but so far no one has show PoC code for BOOM.

Regardless, it's a fair assumption that everyone working with or on CPU
implementations are knowledgable about this and thinking about it.

Tommy

donggyukim, 2018-01-25:
There's a related thread in Google Group:
https://groups.google.com/forum/#!topic/riscv-boom/yxDwmpjtQrE

In Berkeley, we also think it's a very important problem, so start hunting a solution.

mytbk, 2018-11-13:
Just saw there's a secureboom branch.

I did some experiment today, and found there is cache refill caused by transient instructions. However, I'm not able to make a exploit yet due to the buggy branch predictor.

ccelio, 2018-11-13:
> buggy branch predictor

\o/ Spectre proof!

Unfortunately yes, the BPU needs performance verification work and the secureboom work is still a work-in-progress.

abejgonzalez, 2018-11-13:
Hi,

So I am one of the students currently working on replicating Spectre on BOOM. If you are interested, I have a repo at (https://github.com/abejgonzalez/boom-attacks) that has Spectre v1 working with the DefaultBoomConfig with the GShare BPU (condBranchMispred.c). 

As for the BPU, yes it is buggy, and we are hoping to verify and get it working! But the code in boom-attacks is a quick POC of Spectre-v1 that avoids the issues in the frontend.

mytbk, 2019-01-09:
@jerryz123 What is the current status of secureboom, are you still working on it? I'm going to do some work on securing BOOM from Spectre, so I'd like to know the current work on it.

jerryz123, 2019-01-10:
The secureboom branch is currently outdated. We implemented a prototype "Spectre-resistant" data cache, and its sitting in a private branch currently. We'll try to move it to the public secureboom branch soon.

mytbk, 2019-01-11:
@jerryz123 would you mind telling me what scheme the new Spectre-resistant cache based on? From what I have read, there are following methods:

- speculative buffer like InvisiSpec, it looks like the current secureboom also uses this scheme
- secure cache partitioning like DAWG
- other methods such as taint tracking

jerryz123, 2019-01-11:
Our implementation is comparable to InvisiSpec

mytbk, 2019-02-26:
I saw the report about secureboom and boom-attack at https://people.eecs.berkeley.edu/~kubitron/courses/cs262a-F18/projects/reports/project4_report.pdf
Is your new implementation able to run SPEC CPU benchmarks now?

abejgonzalez, 2019-02-27:
Unfortunately, since then our team has not been able to work on the project extensively. We hope to get it done soon but as of right now there is no ETA.

Ziqiao1992, 2020-01-08:
> I saw the report about secureboom and boom-attack at https://people.eecs.berkeley.edu/~kubitron/courses/cs262a-F18/projects/reports/project4_report.pdf
> Is your new implementation able to run SPEC CPU benchmarks now?

I am wondering where I can find the implementation for spectre buffer proposed in https://people.eecs.berkeley.edu/~kubitron/courses/cs262a-F18/projects/reports/project4_report.pdf.

ziqiaozhou, 2020-01-08:
I am wondering whether I can find a branch implementing the speculative buffer in 
https://people.eecs.berkeley.edu/~kubitron/courses/cs262a-F18/projects/reports/project4_report.pdf. It seems that secureboom has some similar implementation. However, it still has a cache refill problem due to speculative execution. Is the secureboom branch a complete but outdated branch(works if using old rocket) or an ongoing branch (can not work for now)?

jerryz123, 2020-01-08:
The secureboom branch is stale, and it will be challenging to make it work again (you'd have to track down old versions of several dependencies). I paused work on secureboom to rewrite the L1 refill mechanism to make it more amenable to implementation of a speculation buffer.

The good news is that it should be straightforward to implement this feature on your own now. In src/main/scala/lsu/mshrs.scala, you can see that the refill state machine has been modified such that there is a `drain_rpq_load` state which responds to loads out of the line-fill-buffers, before the refilled line has been "committed" into the data and tag arrays. Currently all loads are forwarded out of the line fill buffer, and then the line is committed into the arrays. It should be ~10 lines of code to alter the behavior, such that the line is only committed if some load to that line was non-speculative.