riscv-boom/riscv-boom #58: Reset $D tag_array

pentin-as, 2018-02-05:
When I ran the same test without RANDOMIZE, I got the following results.
1621: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
277134: mcycle = **10508**
277450: minstret = 4159
 ... succeed
281175: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
549841: mcycle = **10537**
550158: minstret = 4159
 ... succeed
553882: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
822548: mcycle = **10537**
822865: minstret = 4159
 ... succeed
The first case is the beginning of the simulation, the second and third after the reset.
As you can see for the first time the test is performed faster. When I change the original value of $D tag_array, the values ​​of mcycle change.

Obviously, the problem is that the validity of $D is stored in RAM. The problem is also that the reset signal does not come to this RAM. It seems to me that the sign of validity should be in the registers.

pentin-as, 2018-02-06:
I realized that I'm wrong.
tag_array is cleared in the first nSets of cycles after the reset is cleared. It can be seen here:
[image]
This is done here:
https://github.com/freechipsproject/rocket-chip/blob/c1eb795abafc3774c69560bbe3b0cff5210bdd21/src/main/scala/rocket/HellaCache.scala#L254
https://github.com/freechipsproject/rocket-chip/blob/c1eb795abafc3774c69560bbe3b0cff5210bdd21/src/main/scala/rocket/HellaCache.scala#L259
https://github.com/freechipsproject/rocket-chip/blob/c1eb795abafc3774c69560bbe3b0cff5210bdd21/src/main/scala/rocket/HellaCache.scala#L264

But I do not understand why when RANDOMIZE is disabled (random only for RAM) there is a runtime dependence on the random seed.
I see that in one case, the cache miss occurs during the test. In another - after completion. Because of this, times are different.
I will continue to understand.

pentin-as, 2018-02-07:
It seems that branch predictors influence.

After the beginning of the simulation:
Cyc=505
saq[16]=(VAD----) b:0000 0x02a09 -> 0x0000000000000000 H E C  
saq[17]=(V-----) b:0002 0xxxxxx -> 0xxxxxxxxxxxxxxxxx        
saq[18]=(------) b:0000 0xxxxxx -> 0xxxxxxxxxxxxxxxxx       T
stq_tail=18, st_enq_idx=19
Cyc=506
saq[16]=(VAD----) b:0000 0x02a09 -> 0x0000000000000000 H E C  
saq[17]=(V-----) b:0000 0xxxxxx -> 0xxxxxxxxxxxxxxxxx        
saq[18]=(V-----) b:0004 0xxxxxx -> 0xxxxxxxxxxxxxxxxx        
saq[19]=(---) b:0000 0xxxxxx -> 0xxxxxxxxxxxxxxxxx       T

After reset:
Cyc=505
saq[16]=(VAD----) b:0000 0x02a09 -> 0x0000000000000000 H E C  
saq[17]=(V------) b:0007 0x228e8 -> 0x0000000080022958        
saq[18]=(-------) b:0000 0x22958 -> 0x000000000000003d       T
stq_tail=18, st_enq_idx=18
io.brinfo.stq_idx=17, io.brinfo.valid=1, io.brinfo.mispredict=1, io.exception=0
Cyc=506
saq[16]=(VAD----) b:0000 0x02a09 -> 0x0000000000000000 H E C  
saq[17]=(-------) b:0000 0x228e8 -> 0x0000000080022958       T

I will continue to understand.

ccelio, 2018-02-07:
The bimodal predictor bits are randomized and thus could dictate whether a branch is taken or not-taken so maybe that is the source of your differing execution times.

I intend in the future to write a finite state machine to zero-out the branch predictor memories which will add determinism to the simulation and hopefully remove the reliance on random value injection for 4-state simulation.

pentin-as, 2018-02-08:
It turned out that this is the influence of BPD.
The signal "bpd_stage.io.f2_bpd_resp.bits.takens" went to "fetch_unit.io.resp.bits.bpu_info(3).bpd_blame", and then to "lsu.io.brinfo.bpd_mispredict". This caused the sdq queue to change. And this changed the execution time.

The results if zeroed p_table and h_table.
1621: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
272382: mcycle = **10580**
272698: minstret = 4159
 ... succeed
276423: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
547191: mcycle = **10580**
547508: minstret = 4159
 ... succeed
551233: /home/pentin-as/crypt/titan_lfs/sim/data/gen/median.riscv started
822001: mcycle = **10580**
822318: minstret = 4159
 ... succeed

pentin-as, 2018-10-29:
I think the problem is solved in https://github.com/ucb-bar/riscv-boom/pull/86.

closed 2018-10-29 by pentin-as