riscv-boom/riscv-boom #70: Unrecognized VM option 'MaxPermSize=256M'

nihilscire, 2018-04-12:
I've been doing `make CONFIG=BoomConfig` and this is the result:
```
make -C /***/boom-template/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /***/boom-template/rocket-chip/sbt-launch.jar" root_dir=/***/boom-template/rocket-chip/firrtl build-scala
make[1]: Entering directory '/***/boom-template/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /***/boom-template/rocket-chip/sbt-launch.jar "assembly"
Unrecognized VM option 'MaxPermSize=256M'
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
Makefile:25: recipe for target '/***/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar' failed
make[1]: *** [/***/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 1
make[1]: Leaving directory '/***/boom-template/rocket-chip/firrtl'
/***/boom-template/Makefrag:24: recipe for target '/***/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar' failed
make: *** [/***/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 2
```
Java version:
```
$ java -version
openjdk version "9-internal"
OpenJDK Runtime Environment (build 9-internal+0-2016-04-14-195246.buildd.src)
OpenJDK 64-Bit Server VM (build 9-internal+0-2016-04-14-195246.buildd.src, mixed mode)
```

This problem arises because MaxPermSize has been removed since Java 8.
I'll leave some links below:
[http://openjdk.java.net/jeps/122](http://openjdk.java.net/jeps/122)
[http://www.oracle.com/technetwork/java/javase/8-compatibility-guide-2156366.html](http://www.oracle.com/technetwork/java/javase/8-compatibility-guide-2156366.html)

ccelio, 2018-04-12:
Does everything work if (https://github.com/ccelio/boom-template/blob/master/Makefrag-variables#L20) is changed to remove that `MaxPermSize` option?

I'll remove that option.

In general, I'm wary of Java 9 (which is no longer supported, whereas Java 8 has Long Term Support), so let me know if you get it working.

nihilscire, 2018-04-17:
Even if I remove that option, it doesn't compile at all. This is due to Java 9, so I decided to switch to Java 8. Now it works well. 

Idea: open a new issue with the purpose of writing a requirements/prerequisites document. (If it exists, I did not find it, sorry)

You may close this issue, thanks for your help.

jackkoenig, 2018-04-17:
Wow I didn't realize that. Perhaps we could check the version and error in Makefiles or even in sbt

ccelio, 2018-04-20:
Fixed by https://github.com/ccelio/boom-template/commit/993c6a7f50e5b0d1fb09454e608b6ba90591b02f

closed 2018-04-20 by ccelio