riscv-boom/riscv-boom #87: The riscv-boom does not compile with the last rocket

pentin-as, 2018-06-01:
rocket-chip version: 4269a22

Error text:
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error] at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:51)
[error] at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:48)
[error] at chisel3.core.Module$.do_apply(Module.scala:49)
[error] at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:297)
[error] at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:295)
[error] at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] at chisel3.internal.Builder$.build(Builder.scala:295)
[error] at chisel3.Driver$.elaborate(Driver.scala:93)
[error] at freechips.rocketchip.util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:54)
[error] at boom.system.Generator$.elaborate(Generator.scala:13)
[error] at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:91)
[error] at boom.system.Generator$.circuit$lzycompute(Generator.scala:13)
[error] at boom.system.Generator$.circuit(Generator.scala:13)
[error] at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:97)
[error] at boom.system.Generator$.generateFirrtl(Generator.scala:13)
[error] at boom.system.Generator$.delayedEndpoint$boom$system$Generator$1(Generator.scala:95)
[error] at boom.system.Generator$delayedInit$body.apply(Generator.scala:13)
[error] at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] at scala.collection.immutable.List.foreach(List.scala:381)
[error] at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] at scala.App$class.main(App.scala:76)
[error] at boom.system.Generator$.main(Generator.scala:13)
[error] at boom.system.Generator.main(Generator.scala)
[error] at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] at java.lang.reflect.Method.invoke(Method.java:498)
[error] at sbt.Run.invokeMain(Run.scala:93)
[error] at sbt.Run.run0(Run.scala:87)
[error] at sbt.Run.execute$1(Run.scala:65)
[error] at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.OutOfMemoryError: GC overhead limit exceeded
[error] at scala.collection.mutable.ListBuffer.$plus$eq(ListBuffer.scala:174)
[error] at scala.collection.mutable.ListBuffer.$plus$eq(ListBuffer.scala:45)
[error] at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] at scala.collection.immutable.List.foreach(List.scala:381)
[error] at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
[error] at scala.collection.immutable.List.map(List.scala:285)
[error] at freechips.rocketchip.rocket.Simplify$$anonfun$21$$anonfun$apply$33.apply(Decode.scala:131)
[error] at freechips.rocketchip.rocket.Simplify$$anonfun$21$$anonfun$apply$33.apply(Decode.scala:131)
[error] at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[error] at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[error] at scala.collection.immutable.List.foreach(List.scala:381)
[error] at scala.collection.TraversableLike$class.flatMap(TraversableLike.scala:252)
[error] at scala.collection.immutable.List.flatMap(List.scala:344)
[error] at freechips.rocketchip.rocket.Simplify$$anonfun$21.apply(Decode.scala:131)
[error] at freechips.rocketchip.rocket.Simplify$$anonfun$21.apply(Decode.scala:131)
[error] at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
[error] at scala.collection.immutable.List.foldLeft(List.scala:84)
[error] at freechips.rocketchip.rocket.Simplify$.getCover(Decode.scala:131)
[error] at freechips.rocketchip.rocket.Simplify$.apply(Decode.scala:142)
[error] at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$7.apply(Decode.scala:39)
[error] at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$7.apply(Decode.scala:29)
[error] at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] at scala.collection.immutable.Range.foreach(Range.scala:166)
[error] at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
[error] at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error] at freechips.rocketchip.rocket.DecodeLogic$.apply(Decode.scala:29)
[error] at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$13.apply(Decode.scala:50)
[error] at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$13.apply(Decode.scala:49)
[error] at scala.collection.TraversableLike$WithFilter$$anonfun$map$2.apply(TraversableLike.scala:728)
[error] at scala.collection.immutable.List.foreach(List.scala:381)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] at sbt.Run$.executeTrapExit(Run.scala:124)
[error] at sbt.Run.run(Run.scala:77)
[error] at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] at scala.util.Try$.apply(Try.scala:209)
[error] at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 210 s, completed May 31, 2018 4:55:30 PM

All is compiled if I cancel the following corrections:
https://github.com/freechipsproject/rocket-chip/commit/492ac75ecf2454859dc97be6b6473111df1586ff#diff-61210e44ff43865a122adf453037b4c6R102 - 106
https://github.com/freechipsproject/rocket-chip/commit/492ac75ecf2454859dc97be6b6473111df1586ff#diff-61210e44ff43865a122adf453037b4c6R130 - 131

All is compiled if I remove the following DecodeLogic call in BOOM:
https://github.com/ucb-bar/riscv-boom/blob/aaf42c950cd1620d7b6c4e6cc22b43aa306f3739/src/main/scala/exu/fudecode.scala#L38

pentin-as, 2018-06-01:
https://github.com/pentin-as/riscv-boom/tree/rocket_may2018

ccelio, 2018-06-01:
```bash
[error] at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.OutOfMemoryError: GC overhead limit exceeded
[error] at scala.collection.mutable.ListBuffer.$plus$eq(ListBuffer.scala:174)
```

From your error message, it appears you are running out of memory? Does trying to give your java more memory help?

It's not entirely clear what we can do about that if upstream rocket-chip wants to use their cover stuff here. Perhaps one could refactor the ALU's rrd stage decoder to use less logic? Break it into two pieces? Yuck.

codelec, 2018-06-02:
adding these two lines here https://github.com/esperantotech/boom-template/blob/master/Makefrag#L12   resolves the error 
```bash 
sed -i 's/ getCover(prime2, uncovered, bits)/ getCover(prime2, Seq[Term](), bits)/g' ./Decode.scala
sed -i 's/minterms.foreach(t => assert(cover.exists(_.covers(t))))//g' Decode.scala
```
but this like temporary resolution

pentin-as, 2018-06-04:
var dec_table = AluRRdDecode.table
   if (supported_units.bru) dec_table ++= BruRRdDecode.table
   if (supported_units.mem) dec_table ++= MemRRdDecode.table
   if (supported_units.muld) dec_table ++= MulDivRRdDecode.table
   if (supported_units.csr) dec_table ++= CsrRRdDecode.table
   //if (supported_units.fpu) dec_table ++= FpuRRdDecode.table
   if (supported_units.fdiv) dec_table ++= FDivRRdDecode.table
   if (supported_units.ifpu) dec_table ++= IfmvRRdDecode.table
   val rrd_cs = Wire(new RRdCtrlSigs()).decode(io.rrd_uop.uopc, dec_table)
-Xmx2G
It works

   var dec_table = AluRRdDecode.table
   //if (supported_units.bru) dec_table ++= BruRRdDecode.table
   //if (supported_units.mem) dec_table ++= MemRRdDecode.table
   //if (supported_units.muld) dec_table ++= MulDivRRdDecode.table
   //if (supported_units.csr) dec_table ++= CsrRRdDecode.table
   if (supported_units.fpu) dec_table ++= FpuRRdDecode.table
   //if (supported_units.fdiv) dec_table ++= FDivRRdDecode.table
   //if (supported_units.ifpu) dec_table ++= IfmvRRdDecode.table
   val rrd_cs = Wire(new RRdCtrlSigs()).decode(io.rrd_uop.uopc, dec_table)
-Xmx8G
Out Of Memory

pentin-as, 2018-06-04:
object FpuRRdDecode extends RRdDecodeConstants
{
   val table: Array[(BitPat, List[BitPat])] =
              Array[(BitPat, List[BitPat])](
                               // br type
                               // |      use alu pipe              op1 sel   op2 sel
                               // |      |  use muldiv pipe        |         |         immsel       csr_cmd
                               // |      |  |  use mem pipe        |         |         |     rf wen |
                               // |      |  |  |  alu fcn  wd/word?|         |         |     |      |
                               // |      |  |  |  |        |       |         |         |     |      |
         BitPat(uopFCLASS_S)->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCLASS_D)->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

//         BitPat(uopFMV_S_X)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
//         BitPat(uopFMV_D_X)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMV_X_S)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMV_X_D)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFSGNJ_S)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFSGNJ_D)->List(BR_N , Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

         BitPat(uopFCVT_S_D) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_D_S) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

// TODO comment out I2F instructions.
         BitPat(uopFCVT_S_W) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_S_WU)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_S_L) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_S_LU)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_D_W) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_D_WU)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_D_L) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_D_LU)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

         BitPat(uopFCVT_W_S) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_WU_S)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_L_S) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_LU_S)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_W_D) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_WU_D)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_L_D) ->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFCVT_LU_D)->List(BR_N,Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

/*        BitPat(uopFEQ_S)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFLT_S)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFLE_S)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFEQ_D)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFLT_D)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFLE_D)   ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

         BitPat(uopFMIN_S)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMAX_S)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMIN_D)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMAX_D)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

         BitPat(uopFADD_S)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFSUB_S)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMUL_S)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFADD_D)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFSUB_D)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMUL_D)  ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),

         BitPat(uopFMADD_S) ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMSUB_S) ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
[... cut ...]
         BitPat(uopFNMSUB_S)->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMADD_D) ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFMSUB_D) ->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),
         BitPat(uopFNMADD_D)->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N),*/
         BitPat(uopFNMSUB_D)->List(BR_N, Y, N, N, FN_X   , DW_X  , OP1_X   , OP2_X   , IS_X, REN_1, CSR.N))
}

It works

ccelio, 2018-06-04:
Is it possible to run the Chisel compile through a Scala analysis tool to see how badly the memory usage is between working and broken scenarios? I remember doing this once but I forgot what I used.

Maybe we can split the FPU table into separate single-precision and double-precision to fix it?

pentin-as, 2018-06-05:
// BitPat(uopFMIN_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMAX_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
This is compiled

BitPat(uopFMIN_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
// BitPat(uopFMAX_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
This is compiled

val uopFMAX_S    = UInt(101,UOPC_SZ)
val uopFMIN_D    = UInt(**124**,UOPC_SZ)
val uopFMAX_D   = UInt(**125**,UOPC_SZ)
val uopFADD_S    = UInt(104,UOPC_SZ)
This is compiled

But:
//BitPat(uopFMIN_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
//BitPat(uopFMAX_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMIN_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMAX_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFADD_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFSUB_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
Out Of Memory

BitPat(uopFMIN_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMAX_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMIN_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
BitPat(uopFMAX_D)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
//BitPat(uopFADD_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
//BitPat(uopFSUB_S)  ->List(BR_N, Y, N, N, FN_X, DW_X, OP1_X, OP2_X, IS_X, REN_1, CSR.N),
Out Of Memory

If it were not for the last two experiments, I would suggest that the problem of Out Of Memory is solved when a hole appears in the table.
But the last two experiments suggest that there is an error in FMIN_D and FMAX_D.
What do you think?

Can for now leave a variant with a hole in the table and look for an error at the next incident?

ccelio, 2018-06-05:
My fear is that `rocket.Decode` is implementing a Quine-McCluskey algorithm which is NP-hard. It's possible that our slightly bigger table hits the wrong part of a growing exponential and causes a failure. In the past, this caused me to break up the tables into smaller chunks. 

(https://en.wikipedia.org/wiki/Quine–McCluskey_algorithm)

jackkoenig, 2018-06-05:
I believe it is implementing Quine-McCluskey, @aswaterman can correct.

I've toyed with the idea of integrating espresso into rocket-chip ([here's a version with modern C](https://github.com/galengold/espresso-logic)). Unfortunately, afaik there is no implementation of the espresso algorithm in Scala nor Java. There are ways to convert C code to Java or compile C code to Java class files. In any case it might be worth biting this bullet at some point.

pentin-as, 2018-06-06:
In the patch variant, where I combined some uopXXXX signals, which, with the outputs of all DecodeLogic, had the same result.
What do you think about this change?
[simplify_fpu_decoder.zip](https://github.com/ucb-bar/riscv-boom/files/2075430/simplify_fpu_decoder.zip)

ccelio, 2018-06-08:
@pentin-as Huh, now I'm confused at how the FPU discerns between different ops, for example MIN and MAX are now combined! My head hurts. I guess it's getting fed surreptitiously through the `imm_packed` field?

pentin-as, 2018-06-08:
The proposed code is compiled, but does not work. At least the FADD.S test does not work. So far I have not had time to figure out why.
I was also embarrassed that the instructions FMIN.S and FMAX.S (and several others) are the same. But nevertheless, the decoder simplifies.
I wanted to ask whether it is possible to simplify the decoder in this way (it may be planned, add differences in the future)?
P.S. I'll try to find the time to find the reason why the tests do not work and fix it.

pentin-as, 2018-06-17:
Compiled, but the tests fell because of my error. Commented on FADD.S and FSUB.S instructions in fudecode.scala :). When the error was fixed, Out-Of-Memory appeared again.
It seems that simplification of the decoder does not give anything, so I'm not sure if it's worth dividing into two phases ...
But the problem of Out-Of-Memory disappears if I insert a hole in the instruction codes, like this:
val uopFMINMAX_D = UInt (85, UOPC_SZ)
 // = UInt (86, UOPC_SZ)
val uopFADD_S = UInt (87, UOPC_SZ)
While sending the merge request of the current working version with the simplified decoder (so that at least integration with the last rocket would be):
https://github.com/ucb-bar/riscv-boom/pull/90
I have no idea what to do yet. It is possible to really use the analysis tool for scala ...

ccelio, 2018-06-18:
As scala is really just invoking the java JVM, I believe you can modify the Makefile (https://github.com/esperantotech/boom-template/blob/master/Makefrag-variables#L20)'s invocation of java to add in instrumentation/analysis (java analysis tools shouldn't be too hard to find).

I can confirm I've reproduced the problem reported by others in this thread. I'll see if I can demo a fix to lesson the load on Quine-McCluskey. :/

ccelio, 2018-06-19:
I did I few of my own experiments on this.  We can combine a number of the FP micro-ops together. Fmin/fmax, fcvt_x_s, fcvt_x_d, etc. 

However, it seemed that the value of the uopcode changed whether or not the compilation hits an OOM. I have no intuition on why changing the values of the keys going into the rocket.Decode algorithm changes memory usage and run-time, but it appears to do so.