riscv-boom/riscv-boom #92: Travis CI?

jerryz123, 2018-06-21:
Has there been any work done towards enabling Travis for BOOM, similar to what is done for RocketChip?

ccelio, 2018-06-21:
Nope. Feel free to take a stab at it though.

There are a couple of challenges here. 

First, ideally the Travis would gate commits to riscv-boom, but riscv-boom is not a standalone repository. So any changes that require a synchronized update across dependent repositories (boom-template, rocket-chip, riscv-tools) would not work for this type of CI flow.

There is also is the run/build time. Travis often timed out on rocket, and rocket is relatively simply and runs much faster than boom. There's also the hassle of trying to get it to cache the riscv-tools to improve run time, instead of rebuilding it every time.

Watching from afar, it seemed like a headache for freechips/sifive to deal with, so I never gave it a go.

aswaterman, 2018-06-21:
Travis nodes are also limited to 8 GiB of memory.

ccelio, 2018-06-21:
Ouch, thanks for the additional info @aswaterman .

jerryz123, 2018-06-21:
Thanks for the info, I'm trying to figure out if this is feasible before I try.

> First, ideally the Travis would gate commits to riscv-boom, but riscv-boom is not a standalone repository.

Couldn't we make the Travis script fetch some stable release of boom-template and its submodules, and the current commit of BOOM?

> There is also is the run/build time. Travis often timed out on rocket, 

It seems like Travis has no timeout limit, assuming every job outputs something every 10 minutes. 

> Travis nodes are also limited to 8 GiB of memory.

Is 8 GB of memory not enough for the build? I'm a little more concerned about the disk space. which is capped at 22 GB for the Ubuntu 12.04 image. My boom-template folder right now is 20 GB.

ccelio, 2018-06-21:
The problem is when you need to make a change across rocket-chip+riscv-boom+boom-template, there won't yet exist the proper boom-template to test against, since the leaf repos have to be updated first. The only thing you can do is pull out your sudo permissions and override the Travis failure gating off commits.

I thought I saw that boom takes ~13 GB of VM during Chisel compile, but maybe I'm wrong about that. Your folder should only be ~1 GB, unless you are storing *.vpd files or you keep around the riscv-tools build.

ccelio, 2018-06-21:
For chisel->firrtl compile, top shows mostly one core at 150-200% CPU, 2.2% MEM, 11.7G VIRT, 1.3 GB RES.

Not entirely sure how "fake" VIRT is ("virtual size of a process"), but that's where I saw that "13GB".

firrtl->verilog compile, top shows 400% CPU at 2.7% MEM (1.6 GB RES, 6.7 GB VIRT).

codelec, 2018-06-22:
You can try https://github.com/marketplace/circleci 
I am using that to test all possible boom configs(even mega). It has a timeout of 5hrs and for free account a limit of 4 concurrent docker instances though you can also combine 2 docker instances(4 vCPU, 8096MB RAM) from the dashboard. Combining docker instances will be required to simulate MegaBoomConfig for eg build job in 
https://circleci.com/gh/codelec/project-template 
https://github.com/codelec/project-template/blob/circleci/.circleci/config.yml 
@ccelio we can set that up for esperantotech/boom-template
* by combining -> i mean at high level it seems like that

closed 2018-09-26 by jerryz123