riscv-boom/riscv-boom #93: pipeview: how to generate custom *.out files

abdullahyildiz, 2018-07-09:
Sorry for this NOOB question.

I can run an existing assembly test and then view the corresponding pipeline states by using the generated *.out file:
```
make output/rv64ui-p-simple.out
../boom/util/pipeview-helper.py -f output/rv64ui-p-simple.out > clean_trace.out
../../gem5/util/o3-pipeview.py --color --store_completions -o pipeview.out clean_trace.out
less -r pipeview.out
```
How can I generate a custom *.out file for my assembly code?

ccelio, 2018-07-09:
When you type `make output/rv64ui-p-simple.out` it invokes your emulator on the `output/rv64ui-p-simple` RISC-V binary. You can see the commands it uses and replace the `output/rv64ui-p-simple` binary with your own desired RISC-V binary.

pentin-as, 2018-07-09:
in boom-template/verisim:
./simulator-boom.system-DefaultBoomConfig <YOUR_BINARY_FILE> > <YOUR_FILE_OUT.out>

closed 2018-07-09 by abdullahyildiz