riscv-boom/riscv-boom #98: LSU: Assertion hit that stq is inconsistent

pranith, 2018-07-30:
I am hitting the assertion that stq is inconsistent sometimes when running dhrystone. I managed to capture the signals on one such run.

In the capture, you can see that stq_succeeded_29 is set at the same time when io_nack_cache_nack for the same entry is received. This nack resets the stq_executed_29, but does not touch stq_succeeded_29 causing the stq_head to move past stq_executed_head.

Any idea if this is an issue isolated to the LSU or something is amiss in the dcache?
![boom_inconsistent_stq]

(https://user-images.githubusercontent.com/1878910/43381919-042f3e80-938b-11e8-9c40-02a72def4b29.png)

ccelio, 2018-07-30:
Thanks for the report! What commit hash are you on for riscv-boom? Is this a recent regression? What commands did you type to hit this? Is this the dhrystone from riscv-tests? What is the hash on your riscv-tests?

Can you show the will_fire* signals and the memreq signals in the LSU? (valid, uopc, mem_cmd, laq/stq index). As well as any "kill" signals.

My confusion is that, IIRC, stores are only given a 1 cycle window to be nacked but it appears to receive the nack outside that window. I'm also confused about the `stq_executed_head` pointer. Why is it advancing from 0x10 to 0x11 but `stq_executed_29` goes high?

pranith, 2018-08-03:
Apologies! These results are indeed with my changes to the LSU. I plan to submit them as a PR once the issues are fixed. Until then you can find them here: https://github.com/pranith/riscv-boom/commit/06c09040f7f8b2da09ae4aeacb9ce0ba887efbfd

Seems like after my changes, the NACK always comes after 2 cycles after executing the store. I am new to Chisel and really not sure why this is delay is being introduced. I will try to figure it out.

Please do let me know if you see something obvious that I am missing.

closed 2018-08-03 by pranith