riscv-boom/riscv-boom #100: Boom emulator for Zynq-FPGA

hoangt, 2018-08-03:
I tool the latest Zynq-FPGA for Boom from https://github.com/donggyukim/fpga-zynq. Unfortunately, it does not support emulator for Boom core. Therefore, I imported "verisim" folder and its dependencies from "boom-template" repo (https://github.com/esperantotech/boom-template) and try to make a simulator for Boom (FPGA). With some minor changes, I can build Boom core using Verilator. 

When I run a simple test (i.e. make output/rv64ui-p-simple.out), it works well. However, when I run a benchmark test (i.e. make output/dhrystone.riscv.out), it failed and showed below error

```
simulator-boom.system-BoomConfig  +max-cycles=10000000  +verbose  output/dhrystone.riscv 3>&1 1>&2 2>&3 | /home/hoangt/WORK/TOOLS/risc-v/TOOLCHAIN/boom/boom-zynq/bin/spike-dasm  > output/dhrystone.riscv.out && [ $PIPESTATUS -eq 0 ]
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 35593
[294943] %Error: boom.system.TestHarness.BoomConfig.v:430488: Assertion failed in TOP.TestHarness.ExampleBoomSystem.tile.core
%Error: /home/hoangt/WORK/PROJECTS/RISCV/fpga-hwlog/fpga-boom/verisim/generated-src/boom.system.TestHarness.BoomConfig.v:430488: Verilog $stop

```
Anybody got this issue before and any hint to resolve this.
Thanks
/T

pranith, 2018-08-03:
What does the file output/dhrystone.riscv.out contain?

hoangt, 2018-08-03:
It is log file
```
using random seed 1533342307
Assertion failed: Pipeline has hung.
    at core.scala:1060 assert (!(idle_cycles.value(13)), "Pipeline has hung.")

```

jerryz123, 2018-08-03:
Interesting. Which commit hash of boom are you on?

hoangt, 2018-08-04:
I use Boom #1373dc5d which can be synthesized on Zynq. What would be the compatible version of rocket chip for that boom version? I knew that boom-template in master branch passes dhrystone but I have not been successful to merge zynq-fpga on top or create FPGA package for Boom master

hoangt, 2018-08-06:
Solved based on [https://github.com/ucb-bar/riscv-boom/issues/56](url)

closed 2018-08-06 by hoangt