riscv-boom/riscv-boom #101: False page-fault and access exceptions.

jerryz123, 2018-08-07:
We are seeing a bug with BOOM running user programs in Linux in our Firesim testing. 

```
[    0.996000] xargs[57]: unhandled signal 11 code 0x1 at 0x0000000000000038 in libc-2.26.so[2000024000+13e000]
[    0.996000] CPU: 0 PID: 57 Comm: xargs Not tainted 4.15.0-rc6-31580-g9c3074b5c2cd-dirty #21
[    0.996000] sepc: 0000002000093804 ra : 00000020000937d0 sp : 0000003fffc1da80
[    0.996000]  gp : 00000000000d54c8 tp : 00000020000236d0 t0 : 0000000000000659
[    0.996000]  t1 : 0000002000094958 t2 : ffffffffffffffff s0 : 0000000000000000
[    1.000000]  s1 : 0000000000000888 a0 : 0000000000000000 a1 : 00000000000de270
[    1.000000]  a2 : 00000000000df270 a3 : fffffffffffff7ff a4 : 0000002000092a6c
[    1.000000]  a5 : 00000000000de270 a6 : 00000000000000ff a7 : 0000000000000104
[    1.000000]  s2 : 00000020001648f0 s3 : 0000000000000bd0 s4 : 0000002000164c38
[    1.000000]  s5 : 0000000000000000 s6 : 0000000000000000 s7 : 00000000000b75d8
[    1.000000]  s8 : 00000000000d51b8 s9 : 00000000000077f5 s10: 0000000000000000
[    1.000000]  s11: 0000000000000040 t3 : 000000200009106c t4 : 0000000000000002
[    1.000000]  t5 : 0000002000027368 t6 : 0000000000000000
[    1.000000] sstatus: 8000000200006020 sbadaddr: 0000000000000038 scause: 000000000000000d
```

- The unhandled signals are always 7 (SIGBUS) or 11 (SIGSEGV). 
- The error is extremely sensitive to the state of the machine
- sepc does not always point to a memory instruction, (although I'm not 100% certain I'm reading the right instruction from the sepc)
- sbadaddr is an unreasonable address

Basically, BOOM is generating false access and pagefault exceptions for user code. I haven't been able to reproduce this issue in a smaller test case. Actually, I'm not always able to reproduce this issue at all. 

What's strange to me is what could be setting sbadaddr to such strange values. It seems like an invalid request is being sent through the LSU somehow, but I don't see where this could be possible.

ccelio, 2018-08-07:
Is `scause` consistent? Are you synthesizing asserts when running on the FPGA? I'd guess you're getting a terrible interaction between loads and mispredicted branches.

jerryz123, 2018-08-07:
`scause` is always `0x5, 0x7, 0xd, 0xf`, although I think 90% of the time its `0xd` or `0xf`.

I am not synthesizing asserts on the FPGA. @donggyukim can we do this now?

If it were only due to interactions between loads and branches, then wouldn't we expect to see these show up in kernel code? I've only seen this show up in user code, always in libc or ld, although this is probably because most of the early processes spend most of their time in those libraries.

What's also weird is that this did not show up at all in perlbench, which ran for 7 trillion cycles.

closed 2021-05-01 by jerryz123