riscv-boom/riscv-boom #103: Boom with RoCC

hoangt, 2018-08-13:
Hello,

Does Boom support RoCC interface? I could not generate RoCC emulator in Boom verisim (i.e. similar RoccExampleConfig in rocket-chip/emulator). We use #bdd16b464 (https://github.com/donggyukim/fpga-zynq.git) because we need to have FPGA support.

Anybody has hints to resolve this issue ?
/T

jerryz123, 2018-08-13:
Support for ROCC in BOOM has not been completed.

On page 62 of the BOOM [documentation](https://ccelio.github.io/riscv-boom-doc/), Chris outlines several ways ROCC can be implemented. The simplest method, to just stall decode for every ROCC instruction, should be straight-forward to implement. This method is similar to how BOOM treats CSR instructions, 

The version of BOOM in donggyu's repository is BOOMv1, which is quite old. The current version, BOOMv2, is what we recommend users should try. FPGA synthesis for BOOMv2 is supported through [FireSim](https://fires.im/), an open hardware simulation platform which leverages AWS F1 FPGA instances. This is currently the best way to run BOOM on FPGA.

jerryz123, 2019-07-10:
RoCC support has been completed.

closed 2019-07-10 by jerryz123