riscv-boom/riscv-boom #104: Build fails due to outdated scripts

tampler, 2018-08-15:
Hello
I followed installation instruction and tried to build a small version with:
make PROJECT=test CONFIG=SmallBoomConfig
Build fails due to 

[info] Running test.Generator /home/bku/work/cores/boom-template/verisim/generated-src test TestHarness test SmallBoomConfig
[error] (run-main-0) java.lang.ClassNotFoundException: test.Generator
[error] java.lang.ClassNotFoundException: test.Generator
[error] 	at java.lang.ClassLoader.findClass(ClassLoader.java:530)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
[error] 	at sbt.internal.inc.classpath.ClasspathFilter.loadClass(ClassLoaders.scala:70)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
[error] 	at java.lang.Class.forName0(Native Method)
[error] 	at java.lang.Class.forName(Class.java:348)
[error] 	at sbt.Run.getMainMethod(Run.scala:98)
[error] 	at sbt.Run.run0(Run.scala:86)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed Aug 15, 2018 8:23:20 PM

I'm able to build and simulate a standalone Rocket chip seamlessly. Using Ubuntu 16.04 with javac 1.8.0_181 and all latest sources.

Please help

jerryz123, 2018-08-15:
Try running `make CONFIG=SmallBoomConfig`

tampler, 2018-08-15:
Hi Jerry

Thank you, that worked. However, I see that the code is a bit outdated and should be refactored to support Chisel3 and the latest Rocket. For instance, I see lots of warnings during the build, including:

[deprecated] Math.scala:13 (139 calls): apply is deprecated: "Use log2Ceil instead"
[deprecated] Enum.scala:43 (15 calls): apply is deprecated: "use Enum(n)"
[deprecated] issue_slot.scala:84 (12 calls): apply is deprecated: "Use Reg(t), RegNext(next, [init]) or RegInit([t], init) instead"
[warn] There were 3 deprecated function(s) used. These may stop compiling in a future release, you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate, enable scalac compiler deprecation warnings by either:

Should I create a pull request for this or you already have somebody assigned to this task ? Pentin-as was going to achieve something similar with his pull request

jerryz123, 2018-08-15:
Feel free to PR changes to use Chisel3 syntax. 

I think we'll support the latest rocket-chip once Chris reviews #90 .