riscv-boom/riscv-boom #105: LSU busy-clearing mechanism seems strange

jerryz123, 2018-08-16:
https://github.com/ucb-bar/riscv-boom/blob/ea0ef4e135452f22bbe6abf5d98351bd22c846bc/src/main/scala/lsu/lsu.scala#L726-L729

This mechanism seems strange to me. Is the situation where the LSU will try to clear two separate ROB store entries at once very common? The only case I can think of where this happens is when integer stdata and fp_stdata  are received in the same cycle (integer stdata through the MemExeUnit, and fp stdata through the wire to the fp-pipeline). 

Would it be possible to have the LSU handle only one of the following per cycle?
- int/fp load/store addr
- int store data
- fp store data

This would reduce the number of write ports into the the ROB arrays.

jerryz123, 2018-08-16:
Ah right, it wouldn't even be possible to disambiguate between addrs and stdata for integer ops anyways, since they arrive together in BOOM.

The issue really is that I don't see the case of resolving an integer store and a fp store at once as very frequent. 

A better solution could be to disallow accepting both an integer stdata and a fp stdata in the same cycle.

ccelio, 2018-08-16:
You can't eventually avoid the structural hazard as the int store data and fp store data come from different issue queues. The best you could do is buffer the wakeup and eventually back pressure one of the IQs once the buffer fills but that's probably too complex. 

I'm not sure having two wakeup ports here to the rob is probelematic. The timing of when the busy-clear gets received is somewhat lax and the receiving side is only trying to clear a bit. No expensive CAM ports or moving data around.

jerryz123, 2018-08-16:
I guess this is fine for now then, although it does make the vector-supporting LSU a little more verbose, since without buffering fp stdata there needs to be three sets of signals for clearing stores (int/fp/vec).

https://github.com/ucb-bar/riscv-boom/blob/3011a293f8c07ee93c839fa5752ae75586e01e9f/src/main/scala/lsu/lsu.scala#L864-L879

Also in the current vector design ROB entries track a busy bit and a eidx field, which gets incremented by successive vector wakeups. I think the eidx field isn't strictly necessary, but for now its useful for debugging, and simplifies the wakeup network.

closed 2018-08-19 by jerryz123