riscv-boom/riscv-boom #109: Boom 0.8 compile errors on RHEL 6.10 with R 3.5 and g++ 4.7.7

bandired, 2018-09-04:
I am trying to install Boom 0.8 on Redhat Ent Linux 6.10 with R 3.5 and g++ 4.7.7. I have BH and boost installed but when I try to compile Boom I get these errors (see below). I checked for "atomic" and it exists under BH/boost/. I tried include the full path for atomic as an option on g++ compile options and also several other ways but keep running into these errors. Checked CRAN and the check on CRAN indicates similar or other errors for Linux on x86_x64.  

I haven't programmed in C/C++ in over 15 years...please forgive me if I am missing something basic. 

Appreciate all help. Thanks

***********************************
> install.packages("Boom")
Installing package into ‘/usr/lib64/R/library’
(as ‘lib’ is unspecified)
trying URL 'https://cloud.r-project.org/src/contrib/Boom_0.8.tar.gz'
Content type 'application/x-gzip' length 1216979 bytes (1.2 MB)
==================================================
downloaded 1.2 MB

* installing *source* package ‘Boom’ ...
** package ‘Boom’ successfully unpacked and MD5 sums checked
** libs
g++ -m64 -std=gnu++0x -I"/usr/include/R" -DNDEBUG -I. -I../inst/include -IBmath -Imath/cephes  -I"/usr/lib64/R/library/BH/include" -I"/usr/lib64/R/library/RcppEigen/include" -I/usr/local/include   -fpic  -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -c boom_r_tools.cpp -o boom_r_tools.o
In file included from ../inst/include/Models/DataTypes.hpp:36,
                 from ../inst/include/Models/CategoricalData.hpp:25,
                 from ../inst/include/r_interface/boom_r_tools.hpp:30,
                 from boom_r_tools.cpp:22:
../inst/include/cpputil/RefCounted.hpp:24:18: error: atomic: No such file or directory
In file included from ../inst/include/LinAlg/Vector.hpp:31,
                 from ../inst/include/r_interface/boom_r_tools.hpp:24,
                 from boom_r_tools.cpp:22:
../inst/include/cpputil/math_utils.hpp:31: error: expected constructor, destructor, or type conversion before ‘double’
../inst/include/cpputil/math_utils.hpp:35: error: expected constructor, destructor, or type conversion before ‘double’
In file included from ../inst/include/r_interface/boom_r_tools.hpp:24,
                 from boom_r_tools.cpp:22:
../inst/include/LinAlg/Vector.hpp:85: error: ‘BOOM::Vector::Vector(BOOM::Vector&&)’ cannot be defaulted
../inst/include/LinAlg/Vector.hpp:101: error: ‘BOOM::Vector& BOOM::Vector::operator=(BOOM::Vector&&)’ cannot be defaulted
../inst/include/LinAlg/Vector.hpp: In function ‘std::pair<double, double> BOOM::min_max(const VECTOR&)’:
../inst/include/LinAlg/Vector.hpp:394: error: there are no arguments to ‘infinity’ that depend on a template parameter, so a declaration of ‘infinity’ must be available
../inst/include/LinAlg/Vector.hpp:394: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
../inst/include/LinAlg/Vector.hpp:395: error: there are no arguments to ‘negative_infinity’ that depend on a template parameter, so a declaration of ‘negative_infinity’ must be available
../inst/include/LinAlg/Vector.hpp:396: error: expected initializer before ‘:’ token
../inst/include/LinAlg/Vector.hpp:400: error: expected primary-expression before ‘return’
../inst/include/LinAlg/Vector.hpp:400: error: expected ‘;’ before ‘return’
../inst/include/LinAlg/Vector.hpp:400: error: expected primary-expression before ‘return’
../inst/include/LinAlg/Vector.hpp:400: error: expected ‘)’ before ‘return’
In file included from ../inst/include/LinAlg/Matrix.hpp:24,
                 from ../inst/include/r_interface/boom_r_tools.hpp:25,
                 from boom_r_tools.cpp:22:
../inst/include/LinAlg/VectorView.hpp: At global scope:
../inst/include/LinAlg/VectorView.hpp:218: error: ‘BOOM::ConstVectorView::ConstVectorView(BOOM::ConstVectorView&&)’ cannot be defaulted
In file included from ../inst/include/r_interface/boom_r_tools.hpp:25,
                 from boom_r_tools.cpp:22:
../inst/include/LinAlg/Matrix.hpp:55: error: ‘BOOM::Matrix::Matrix(BOOM::Matrix&&)’ cannot be defaulted
../inst/include/LinAlg/Matrix.hpp:69: error: ‘BOOM::Matrix& BOOM::Matrix::operator=(BOOM::Matrix&&)’ cannot be defaulted
../inst/include/LinAlg/Matrix.hpp:316: error: expected ‘;’ before ‘override’
In file included from ../inst/include/r_interface/boom_r_tools.hpp:26,
                 from boom_r_tools.cpp:22:
../inst/include/LinAlg/SpdMatrix.hpp:38: error: ‘BOOM::SpdMatrix::SpdMatrix(BOOM::SpdMatrix&&)’ cannot be defaulted
../inst/include/LinAlg/SpdMatrix.hpp:40: error: ‘BOOM::SpdMatrix& BOOM::SpdMatrix::operator=(BOOM::SpdMatrix&&)’ cannot be defaulted
../inst/include/LinAlg/SpdMatrix.hpp:69: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:94: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:100: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:109: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:146: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:148: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:150: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:153: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:155: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:157: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:160: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:162: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:164: error: expected ‘;’ before ‘override’
../inst/include/LinAlg/SpdMatrix.hpp:167: error: expected ‘;’ before ‘override’
[... 275 lines cut ...]
boom_r_tools.cpp: In member function ‘BOOM::CategoricalData BOOM::Factor::to_categorical_data(int) const’:
boom_r_tools.cpp:554: error: invalid abstract return type for member function ‘BOOM::CategoricalData BOOM::Factor::to_categorical_data(int) const’
../inst/include/Models/CategoricalData.hpp:184: note:   since type ‘BOOM::CategoricalData’ has pure virtual functions
boom_r_tools.cpp:555: error: cannot allocate an object of abstract type ‘BOOM::CategoricalData’
../inst/include/Models/CategoricalData.hpp:184: note:   since type ‘BOOM::CategoricalData’ has pure virtual functions
boom_r_tools.cpp: In member function ‘std::vector<BOOM::Ptr<BOOM::CategoricalData, true>, std::allocator<BOOM::Ptr<BOOM::CategoricalData, true> > > BOOM::Factor::vector_of_observations() const’:
boom_r_tools.cpp:562: error: cannot allocate an object of abstract type ‘BOOM::CategoricalData’
../inst/include/Models/CategoricalData.hpp:184: note:   since type ‘BOOM::CategoricalData’ has pure virtual functions
In file included from ../inst/include/r_interface/boom_r_tools.hpp:28,
                 from boom_r_tools.cpp:22:
../inst/include/LinAlg/Array.hpp: In member function ‘BOOM::Array& BOOM::Array::assign(FwdIt, FwdIt) [with FwdIt = BOOM::VectorViewConstIterator]’:
boom_r_tools.cpp:289:   instantiated from here
../inst/include/LinAlg/Array.hpp:308: warning: comparison between signed and unsigned integer expressions
make: *** [boom_r_tools.o] Error 1
ERROR: compilation failed for package ‘Boom’
* removing ‘/usr/lib64/R/library/Boom’

The downloaded source packages are in
        ‘/usr/sap/tmp/RtmpYCwzGV/downloaded_packages’
Updating HTML index of packages in '.Library'
Making 'packages.html' ... done
Warning message:
In install.packages("Boom") :
  installation of package ‘Boom’ had non-zero exit status
>

jerryz123, 2018-09-04:
I think you are dealing with a different "BOOM" than the one in this repository. This is the Berkeley Out of Order Machine, not the BOOM C++ library for Bayesian modeling.

bandired, 2018-09-05:
Never mind....Thanks Jerry.

closed 2018-09-05 by bandired