riscv-boom/riscv-boom #111: "unhandlable trap 2"

pranith, 2018-09-06:
Trying to run a MegaBoom config is giving me this errors:

../machine/mtrap.c:18: machine mode: unhandlable trap 2 @ 0x0000000080002ff8

Any idea what this might be? The binary works fine with SmallBoom config in firesim.

jerryz123, 2018-09-06:
Hmm could you provide some more detail on how you are compiling and running the binary? Trap 2 is illegal instruction. I'm not sure why this would be thrown if the code works on SmallBoomConfig.

We haven't verified MegaBoomConfig very thoroughly, so we are very interested in any bugs that show up there.

pranith, 2018-09-06:
I should clarify that this was a integer only configuration that you recently committed. My guess is it is using some floating point instructions. I will try to recompile for "rv64ima" version and run it. BTW, when I try compiling as follows : riscv64-unknown-linux-gnu-gcc -march=rv64ima -mabi=lp64... it gives me an error about not finding stubs/lp64.h. Reported on firesim repo: https://github.com/firesim/firesim/issues/91#issue-357787240

pranith, 2018-09-12:
seems to be firesim toolchain issue. closing...

closed 2018-09-12 by pranith