riscv-boom/riscv-boom #116: simulator-boom.system-BoomConfig does not work with proxy kernel

barbara-gigerl, 2018-10-17:
The documentation of simulator-boom.system-BoomConfig tells that it is possible to run an ELF using the proxy kernel. I wrote a simple hello world program for that and compiled it using
`$RISCV/bin/riscv64-unknown-elf-gcc hello.c -o hello`

Then I tried `./simulator-boom.system-BoomConfig pk hello`, just as the documentation says, and I just get

> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1
Listening on port 37739

Then it hangs, obviously listening on port 37739.
The program itself looks like this:
`#include <stdio.h>`
`int main(void){`
  `  printf("Hi from boom\n");`
  `  return 0;`
`}`

Of course I tried it with spike (`spike pk hello`) and it works as expected.

Additionally, I wanted to ask where emulator-freechips.rocketchip.system-SmallBoomConfig mentioned in the readme comes from. I tried `make CONFIG=SmallBoomConfig` in boom-template/rocket-chip/emulator but it says it can't find the config.

Any ideas?

jerryz123, 2018-10-17:
You need to run `make CONFIG=jtagSmallBoomConfig` in `boom-template/verisim`.

`boom-template/rocket-chip/emulator` builds verilator simulations of Rocket configurations, while `boom-template/verisim` builds verilator simulations of BOOM configurations.

jerryz123, 2018-10-17:
Can you run `./simulator-boom.system-jtagSmallBoomConfig +verbose pk hello` and see which exact assertion was fired?

ccelio, 2018-10-17:
@barbara-gigerl the commands you typed should work. It may just take a while to execute and print out your hello world. The jtag output is expected and can be ignored. It simply is providing you another manner to connect to the design under test. You should see the same output when running th regular "make run" regressions.

barbara-gigerl, 2018-10-18:
@jerryz123 When using `+verbose`:
> Using random seed 1539848118 
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1
Listening on port 37739

And then hangs.

When using `--verbose`:
> Using random seed 1539848118 
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1
Listening on port 37739
Assertion failed: Pipeline has hung.
  at score.scala:1107 assert (!(idle_cycles.value(13)), "Pipeline has hung.")
[8257] %Error: boom.system.TestHarness.jtagSmallBoomConfig.v:302381: Assertion failed in TOP.TestHarness.dut.tile.core
%Error: /home/riscv/boom-template/verisim/generated-src/boom.system.TestHarness.jtagSmallBoomConfig.v:302381: Verilog $stop
Aborting...

barbara-gigerl, 2018-10-18:
@ccelio How long is "a while"? The problem is, it is aborting after all these exceptions. 
 `make run` doesn't work with proxy kernel, does it?

jerryz123, 2018-10-18:
@barbara-gigerl I was unable to reproduce the issue using the latest BOOM.

Can you clarify which commit of BOOM you are building?

Also, can you verify if `spike --isa=rv64g hello` works?

barbara-gigerl, 2018-10-19:
@jerryz123 I figured it out, it just takes about 15-20 minutes to compete, just as @ccelio told me. I totally underestimated the time it takes the simulator to execute a simple program like hello world. Thank you very much for your help!

closed 2018-10-19 by barbara-gigerl

phamngocson1408, 2019-07-17:
Hello,

Recently, I have a problem running the simulator. I followed all the steps as follows (of course I have set the $RISCV variable):

```
git clone https://github.com/ccelio/boom-template.git 
cd boom-template
./scripts/init-submodules.sh
./scripts/build-tools.sh
cd verisim
make
```

and then I run comment:

`./simulator-boom.system-BoomConfig +verbose pk hello`

I received the following error:

```
using random seed 1563339559
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 39273
Assertion failed: Pipeline has hung.
    at core.scala:1107 assert (!(idle_cycles.value(13)), "Pipeline has hung.")
[1486095] %Error: boom.system.TestHarness.BoomConfig.v:482231: Assertion failed in TOP.TestHarness.dut.tile.core
%Error: /home/pnson89/tmp/SiFive/BOOM/boom-template/verisim/generated-src/boom.system.TestHarness.BoomConfig.v:482231: Verilog $stop
Aborting...
```

Could you please let me know why this error occured?
Thanks,