riscv-boom/riscv-boom #117: Is the project alive ?

tampler, 2018-10-25:
I see lots of changes and 3 pull requests pending for a review. It looks like Chris is no longer interested in supporting this project. Who else is in charge of reviewing and accepting changes and pushing the design further ?

I'd like to add my changes, but I see some of them have already been implemented, but not yet incorporated in master. How can we push the project? Who will be the boss here?

jerryz123, 2018-10-25:
This project is still very much alive. There just haven't been any critical issues that we've needed to fix. We are working on several large scale projects that will probably get incorporated in the future. Chris is still around to review changes, and we do have a group of students currently working on BOOM. 

What changes are you referring to? We can take a look at these and see if we can incorporate them. We are definitely still interested in continuing to develop this project, and in bringing up a healthy community of users.

tampler, 2018-10-25:
Okay, why don't you review/incorporate/reject dozens of commits from **codelec** and a few nice from **pentin-as** ? What features are you working on? What kind of community support do you need for this? What are microarchitectural changes, compared to the BoomV2 paper? 

The **Dessert** paper refers to some bugs in the latest BOOM frontend. Did you fix those? In the BOOM v2 paper Chris told that the semi-custom register file is likely to be redesigned and memories need to be pipelined... Is that done ? Any help required ?

For me, BOOM currently looks like a bunch of code, generously uploaded by Chris during his PhD research. External contributors wait months for a code review and nobody has an idea what features/changes are under development.

As for useful features, there are lots of Chisel3 migration changes from **codelec**, out-of-memory fix from **pentin-as**, rvc instruction and lots of minor/major changes, which you may see in the "Conflicting files" list for pull requests.

jerryz123, 2018-10-25:
> Okay, why don't you review/incorporate/reject dozens of commits from codelec and a few nice from pentin-as ?

For #97, we really would like to incorporate his changes, but currently his implementation irrevocably changes the frontend. If codelec were to revise his RVC implementation such that it could be configurable, we would be happy to look into integrating it. 

For #86 I don't understand enough to merge this, if @ccelio can verify this we can merge it.

> What features are you working on? What kind of community support do you need for this? What are microarchitectural changes, compared to the BoomV2 paper?

We are going to talk about some of the upcoming work at the Chisel Community Conference in November, a recording of the talk will be uploaded. Some of the major changes we are working on include fixing the memory model, supporting the vector-extension, building a ring-scalar integer pipeline, and cleaning up the codebase.

We are definitely looking for more community support. In the upcoming weeks we are planning to outline a set of open tasks which are a good opportunity for contributors, and a set of guidelines for community contributions. We were planning to present this at the Chisel Community Conference.

> The Dessert paper refers to some bugs in the latest BOOM frontend. Did you fix those?

As far as I'm aware, yes. BOOMv2 now boots Linux successfully on FireSim, as you can see from some of the recent release notes.

> For me, BOOM currently looks like a bunch of code, generously uploaded by Chris during his PhD research. External contributors wait months for a code review and nobody has an idea what features/changes are under development.

We apologize for any lack of communication on our part. We are trying to bring up more people to improve the quality of the codebase and add new features to BOOM. It may be a few months before development gets back in full swing, but rest assured there are many people here interested in supporting and improving BOOM.

ccelio, 2018-10-25:
Hi @tampler, 

As Jerry mentioned, there are a new group of students who are taking on maintenance and improvements to the riscv-boom code base. Although I've graduated, I am still around as a reviewer too.

The best way to contribute big changes is to first coordinate with us on what you would like to do. This prevents you from wasting time on something we may not accept or may conflict with our own internal work, and it helps provide guidance before you start. My second piece of advice is to try and structure changes to be as small and as easy-to-review and easy-to-maintain as possible. That way, if we reject one PR we aren't rejecting all of your contributions.  I'll edit the CONTRIBUTING.md to reflect this advice. 

#97 unfortunately bundles together two separate updates: changes for Chisel3 with a new RVC frontend implementation. The Chisel3 fixes would be accepted, but we cannot accept the RVC frontend. RVC needs to be something that can be disabled, and while interesting, I don't think bypass extra decode stages is the correct technique as it adds extra data path and muxing to the critical path to avoid the uncommon scenario of no-RVC instructions being fetched.

tampler, 2018-10-26:
1. Is there a list of features to work on ? I'd like to pick something if its already available
2. What I want to see in Boom is SMT. I'm interested in working on it. I anticipate some changes in frontend only, but can't estimate the full impact on other parts and software. There should be some micro architectural planning for BOOM and I'd love to talk to those people.
3. What's the best CoreMark and SpecINT2017 score for the current BOOM ? I may also contribute in benchmarking on HW-SW interface.

Is there anybody whom I may talk to ? Or join some discussion group of students and researches @ Berkeley, who are working on this project ? Pls give me a contact email. Thank you!

jerryz123, 2018-10-26:
1. I'm going to clean up the issues list, and add a few new issues/improvements/tasks that can be done. Hopefully then the issues list will be more useful.

2. This sounds like a very interesting, and useful feature. We'd be happy to talk more about this.

3. I'm not sure if we have a recent set of scores. Some numbers were reported at HotChips 2018. We'd really appreciate any community reported performance numbers. 

I think the best place for discussion is the BOOM mailing list. https://groups.google.com/forum/#!forum/riscv-boom

tampler, 2018-10-29:
Thank you! Keep in touch in groups and PRs

closed 2018-10-29 by tampler