riscv-boom/riscv-boom #125: Verilog codes of Boom

Shalaleh, 2018-11-05:
Hello,
I am totally new to riscv-boom. and I need verilog codes of boom.
I did everything at where in the Readme file. Could someone please help me how I can get the Verilog codes of Boom?

Thanks in advance

jerryz123, 2018-11-05:
Running `make` in `boom-template/verisim` will put the generated verilog in `boom-template/verisim/generated-src`

Shalaleh, 2018-11-19:
Thank you.
and if I want to generate Bitstream via Vivado for Riscv-Boom to be implemented on Zedboard, How can I create Constraint file?

Thanks in advance

abejgonzalez, 2019-05-28:
Hi @Shalaleh, has this already been resolved on your end? If so, I will go ahead and close this issue.

Shalaleh, 2019-05-28:
Hello,
No, I gave up at the end.

abejgonzalez, 2019-05-28:
Sorry to hear that. More recently this was released on our end: https://github.com/riscv-boom/fpga-zynq. It might help answer some of your questions. Good luck with future hacking.

closed 2019-05-28 by abejgonzalez