riscv-boom/riscv-boom #128: TAGE predictor

micreven, 2018-11-20:
When I read the source code of scala I have two questions want to ask:
1.Why not use CSR to compress the length of the 90bits global history? 
2.Why use addr to update the global history instead of the speculative prediction？
hope someone can tell me

ccelio, 2018-11-20:
I previously used CSRs for TAGE to get really long history lengths without constant folding and mux-selecting when using a very long register, but found it a ton of state to track and very easy to get it wrong. Each tables required 3 CSRs of ~11 bits each, so that quickly becomes worst than just storing the entire 90 bits.

I don't understand your second question. Which address are we using and which information are you recommending to use?

micreven, 2018-11-20:
> I previously used CSRs for TAGE to get really long history lengths without constant folding and mux-selecting when using a very long register, but found it a ton of state to track and very easy to [...]

Thanks for your answer
In the second question I mean that during the f0 stage you use the function UpdateHistoryHash to update the history and in the function I observe that you fold the io_req_bits_pc and add it to the tail of the old history whose 6th bit to 11th bit was also fold. And my question is what are the advantages of this way to update history compared to add the outcomes of the branch or jump instructions to the tail of the old history ?
And I read the André Seznec, Pierre Michaud's paper about TAGE predictor then I found that the function you use to compute tag is different from the function described in the paper. I want to know why you use indexes to compute tag instead of history and PC.

micreven, 2018-11-22:
Thanks for your answers.

closed 2019-01-27 by ccelio