riscv-boom/riscv-boom #147: CircleCI Improvements

abejgonzalez, 2019-01-06:
Things to improve or look at with the current CI setup

- [ ] cache riscv-tools from boom-template
- [ ] cache verilator build from boom-template
- [ ] save *.out (and other temp. output files) from tests on success and failure
- [ ] build larger configs with more memory (possibly pay for more memory)

jackkoenig, 2019-01-07:
Have ya'll thought about using a docker image for environment stuff like riscv-tools and verilator? We've discussed such an image for Chisel, FIRRTL, and rocket-chip testing, might be good to have a common one for these projects). @ucbjrl might have more thoughts

jerryz123, 2019-01-07:
Yeah a docker image, or even just a curated tarball of riscv-tools would be ideal. It seems like there was a little bit of initiative towards nightly builds a while ago, but that didn't take off. https://groups.google.com/a/groups.riscv.org/forum/#!msg/sw-dev/HdkwlJF4JXA/0bfGtbNiBQAJ

abejgonzalez, 2019-01-07:
@jackkoenig Currently, I have a docker image up at https://cloud.docker.com/u/riscvboom/repository/docker/riscvboom/riscvboom-images that I am currently using for the riscv-boom repo that installs the necessary dependencies needed to build. However, I am not sure if it would be good to package the riscv-tools, chisel, and firrtl things within it since those are project dependent.

jackkoenig, 2019-01-07:
I'm not saying that chisel and firrtl should be packaged, but riscv-tools are pretty stable and you can still have a different docker image for each version and pull the right one in when you need to bump. The issue there is they just take so long to build and at least with Travis, the caching is so flakey it's probably worth putting it in a docker image.

There won't be 1 docker image to rule them all, but automation to build images for new versions that you can point to in a given project would be super useful

abejgonzalez, 2019-01-08:
Sounds like a plan to me. I think a good place to have these images reside would be in the ucbbar organizations docker repo (https://hub.docker.com/u/ucbbar) unless we want to create a new repo somewhere else to store these images (maybe with riscv as the org?). We can also create a repo in the ucbbar github org to store the DockerFiles (or in the riscv github org).

@jackkoenig @ucbjrl Do you happen to know who is in control of the docker hub organization for ucbbar?

ucbjrl, 2019-01-08:
Currently, I'm in control of the ucbbar organization on DockerHub. Shall I put together a team?

abejgonzalez, 2019-01-08:
I think that wouldn't be a bad idea. I already made a DockerFile that installs verilator and riscv tools so I can push the image to a repo there.

As for the DockerFile itself, let me see if I can create a repo on the ucb-bar GitHub and Ill point you and Jack to it.

ucbjrl, 2019-01-08:
There's a private [github repo for the chisel3-tools docker image](https://github.com/ucb-bar/chisel-ci-infra). I've added you as a collaborator so you can check it out.

ucbjrl, 2019-01-08:
@abejgonzalez what's your dockerid, or should I use abe.gonzalez@berkeley.edu?

abejgonzalez, 2019-01-08:
My dockerid is abejgonza. You can add me that way. I am about to PR my DockerFile to https://github.com/ucb-bar/riscv-docker-images so you can also see that too. I am just doing some final checks on things.

ucbjrl, 2019-01-08:
I've added abejgonza as a user/owner to the ucbbar organization on DockerHub.

abejgonzalez, 2019-01-08:
Ok Ive made the first push to https://github.com/ucb-bar/riscv-docker-images. We can move this conversation over there.

pentin-as, 2019-01-09:
Currently, CI is checking the version of boom with the version of rocket-chip corresponding to the latest version of boom-template. This works because the boom-template has not changed since September 2018. If I correct the boom for integration with the latest rocket-chip, the tests will not pass. Moreover, if I update the rocket-chip in the boom-template, they will no longer pass tests of versions of boom that previously worked. I think testing with the latest boom-template is fundamentally wrong.
Ideally (from the point of view of solving this problem) that there was no boom-template repository, and boom had a rocket-chip submodule. But this is not our way, since it is positioned that there should be “your own repository” (above boom) and since with time boom can be integrated not only with rocket-chip.
In this case, the solution may be to create a configuration file, which specifies which version of the rocket-chip should be used for tests. Currently it can be filled with something like:
ROCKET_CHIP_VERSION=50bb13d7887e5f9ca192431234b057ae9d8edb6c

abejgonzalez, 2019-01-10:
This issue came up previously in the last CI issue (#92) where there is a disconnect between what boom needs to build and where you can find it. I see where you are coming from. Having a hash of the rocket-chip version to use would make sense in the boom repo. However, I also see issues with boom-template and boom desyncing (someone might forget to update boom-template at the same time as the rocket-chip hash in boom is getting updated). The people in charge of the org. would have to make sure that both are up to date. I would also think that the majority of people using boom are most likely using boom-template as a starting point (although I have no evidence of this). So having one place (boom-template) keep track of what version of rocket-chip might be convenient to the "average" user. @jerryz123 @ccelio Thoughts?

pentin-as, 2019-01-10:
I would like to add the following thoughts:
  - Duplication of the code is certainly evil. But when building tests, it is much more important that the tests are context independent and reproducible.
  - The main task of the riscv-boom repository tests is that the source of the riscv-boom repository always works. If the more top-level repository (for example, boom-template) is with a wrong version of the rocket-chip, this is already a problem of that repository :)

abejgonzalez, 2019-01-10:
I agree. I will go ahead and make changes. Here are my thoughts...

- Add a hash file stating the version of rocket-chip
- Put in a reference to hashfile it in the README stating that the version of rocket-chip that this version of boom works with
- Update CI to take into account the hashfile when checking out boom-template
- In general, add to the readme and docs the changes for circleci

Thanks for the feedback @pentin-as !

abejgonzalez, 2019-02-27:
Closing since CI seems to work reasonably well right now.

closed 2019-02-27 by abejgonzalez