riscv-boom/riscv-boom #148: Fix RAS

jerryz123, 2019-01-07:
Currently the RAS is left disconnected, yet some of the logic is synthesized when nRAS > 0.
https://github.com/riscv-boom/riscv-boom/blob/f35cc753a9cc5b15402cf7336f8a98912c6533a3/src/main/scala/bpu/bpd-pipeline.scala#L151-L152

This causes the vestigial RAS to inject false predictions to 0x0 in S1 of BTB, sometimes overriding the valid BTB predictions. Preferably the RAS should be fixed and connected.

Side request: Can we also refactor the RAS to be a module, so that the internal registers and io is available to inspect in waveforms?

shinezyy, 2019-05-06:
Is the performance bug mentioned above caused by RAS underflow? I just noticed that medium boom uses only 8 entries of RAS and turns off EmptyCheck. Because calling stack of dhrystone is deep, RAS underflow can degrade perfromance significantly.

jerryz123, 2019-05-06:
We found that for programs with a very shallow call stack, the RAS would mispredict more frequently than expected. There is some issue with the RAS update logic which is polluting it.

closed 2020-04-30 by jerryz123