riscv-boom/riscv-boom #152: BOOM next version TODOs

jerryz123, 2019-01-12:
I think once we get the following features completed, we can mark a new version of BOOM.

- [ ] RVC Support
- [ ] RVWMO memory consistency model
- [ ] PMPs supported
- [ ] Integer-only
- [ ] Configurable RV32 
- [ ] CircleCI
- [ ] PnR head in ROB (Faster CSR instructions)
- [ ] Re-enable RAS
- [ ] RoCC Support
- [ ] Unified issue queues for 1-wide machine (Support BOOMv1 configs again)

abejgonzalez, 2019-01-12:
I would also recommend having HPM's working, TAGE verified and working, and bumped rocket-chip added to the list.

jerryz123, 2019-01-12:
Do we have documentation of the RAS bug anywhere? What is the broken behavior?

How difficult would it be to re-enable HPMs. I see a bunch of commented out code, but I don't know how much work it would be to re-enable these.

Also lets just use the same rocket-chip as whatever FireSim is using. I don't want to bump past the version they have.

abejgonzalez, 2019-01-12:
I don't think HPM's would be that hard. Theoretically, they should already be enabled. It was just when I was trying to read them that I was getting garbage out. As for the RAS, it is just that is it commented out with older signal names. So the port would be to uncomment then fix signal names and verify. As for FireChip, that sounds good with me.

heiner-bauer, 2019-05-15:
After reading chapter 12 of the documentation I have several questions regarding HPMs / HPEs:
- which events does BOOM track ?
  I have no clue of chisel but the code in exu/core.scala https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L221 does not look like it matches the number and ordering of the SiFive U54 events
- how many independent counters does BOOM support ?
- do you happen to have some kind of library to use these counters (configuring and reading these CSRs is kind of fiddly) ?

abejgonzalez, 2019-05-15:
@heiner-bauer Moved the conversation to a new issue thread to keep this thread on "ToDos"

closed 2021-05-01 by jerryz123