riscv-boom/riscv-boom #170: Access exceptions caused by PTW are probably broken right now.

jerryz123, 2019-01-27:
The D$ TLB handles both PTW and core requests, and does not differentiate between either. To re-enable PMPs, #123 ignores access exceptions thrown by the D$ TLB, since core requests check the PMPs in the LSU TLB instead.

However, PTW requests may still fail a PMP check, and that exception needs to propagate.

Essentially, the D$ needs to differentiate between core and PTW requests somehow.

pentin-as, 2019-03-22:
Until you have done more closely L1D and LSU, I would like to try to make the following temporary correction:
 - TLB inside LSU only translate address;
 - D$ contains PMPChecker (instead LSU), which checks all requests (from LSU and PTW);
 - (I'm not sure what happens, but I would like to try) union s1 and s2 clocks inside D$ to read tags and data at the same time (it seems to me that since the addresses in the request are already physical, this is possible and will not increase the critical path... but I'm not sure).

Additionally, I can say that PMP inside LSU contains one of the critical path that would be good to remove:
https://github.com/riscv-boom/riscv-boom/issues/224

What do you think about that?

jerryz123, 2019-03-22:
The issue is this would require forking the TLB and PMPChecker from Rocket, which currently combines them in one module. I'm not sure how much work this would be, and in any case forking would leave behind a lot of duplicate code between BOOM and Rocket. 

It also looks like it will be challenging to split up the functionality of PMPChecker from the TLB, since on refills, the PMPChecker marks bits in the TLB entries.