riscv-boom/riscv-boom #178: Error calculating the address in MegaBoom

pentin-as, 2019-02-01:
MegaBoomConfig

Assertion failed: Pipeline has hung.
    at core.scala:1143 assert (!(idle_cycles.value(13)), "Pipeline has hung.")

This happens as the processor goes to execution from the address 0x10022f88, in which the 'x.

I have the code:
    10001b5c:	64813083          	ld	ra,1608(sp)
    10001b60:	65010113          	addi	sp,sp,1616
    10001b64:	00008067          	ret

sp = 100222f0
When executed, it becomes ra = sp + 1608 + 1616 = 0x10022f88.

This happens because a false bypass happens in iregister_read:
In iregister_read, I see the following:
[image]

This bypass goes from alu to csr_exe_unit:
[image]

It seems to me that this bypass should not be.
I will continue to study the problem. Does anyone have any ideas?

ccelio, 2019-02-01:
Is the test small and reproducible using the default Boom flow? If so, maybe we can take a look.

However, if you have a fsdb file of the failure occurring, then you have everything needed to diagnosis this I think.

Regarding the bypass logic, the only thing that would prevent the `add` from the future bypassing to the `load` behind him is that the `add` should be allocated a new, different physical register specifier (`pdst`). I would check to see what the `pdst` of the bypassed `add` is and compare it to the `pop1` (operand) of the `load`. I assume they will be the same, if this erroneous bypass is occurring.

Then the next step is to figure out who has the wrong physical specifier and why. Did they enter the issue window with the wrong specifier? Did they leave rename with the wrong specifier? Was there a stall in rename? Did the load and add leave rename together? Was there a mispredict and reset?

pentin-as, 2019-02-05:
The problem was that I simulated without RANDOMIZE_MEM_INIT. Because of the false predictions, boom switched to executing the 'x instruction, which led to the breakdown of FetchController and "Pipeline has hung."

Here is a small test that reproduces this situation:

  li   a0, 7
wait:
  addi a0, a0, -1
  bne  a0, x0, wait
  beq  a0, x0, after_wait
kill_boom:
  j    good + 0xfff00 # (at this address must be 'x)
after_wait:
  la   ra, good
  ret

good:
  la   t0, .tohost
  li   t1, 1
  sw   t1, 0(t0)
  j    good

I have a question, should a boom recover from an attempt to execute an 'x instruction as a result of false prediction? I can try to fix it, but I'm afraid of increasing the critical path.

jerryz123, 2019-02-05:
Do you know which module in fetch was the source of the breakdown? I am aware that the fetchbuffer leads to x propagation errors.

jerryz123, 2019-02-05:
I mentioned the fetchbuffer issue here: https://github.com/riscv-boom/riscv-boom/pull/141#discussion_r244556464

abejgonzalez, 2019-05-28:
Hi @pentin-as , has this already been resolved? If so, I will go ahead and close this issue.