riscv-boom/riscv-boom #197: Pipeline Viewer: Incorrect address display after a jump.

pentin-as, 2019-02-19:
MegaBoomConfig

dump:
...
    40001c20:	00050793          	mv	a5,a0
    40001c24:	02d57e63          	bleu	a3,a0,40001c60 <memcpy+0x58>
    40001c28:	00158593          	addi	a1,a1,1
    40001c2c:	fff5c703          	lbu	a4,-1(a1)
    40001c30:	00178793          	addi	a5,a5,1
    40001c34:	fee78fa3          	sb	a4,-1(a5)
    40001c38:	feb618e3          	bne	a2,a1,40001c28 <memcpy+0x20>
    40001c3c:	00008067          	ret
    40001c40:	fed57ee3          	bleu	a3,a0,40001c3c <memcpy+0x34>
    40001c44:	00050793          	mv	a5,a0
...

Pipeline Viewer:
[..............................................................f..dpi....cr......]-(         400000) 0x0040001c40.0 DASM(fed57ee3)            [       351]
[==============================================================f==dpi============]-(         400000) 0x0040001c44.0 -----DASM(00050793)       [       352]
[==============================================================f==dp=i===========]-(         400000) 0x0040001c48.0 -----DASM(00858593)       [       353]
[==============================================================f==dp==i==========]-(         400000) 0x0040001c4c.0 -----DASM(ff85b703)       [       354]
[==============================================================f===dpi===========]-(         400000) 0x0040001c50.0 -----DASM(00878793)       [       355]
[==============================================================f===dp==i=========]-(         400000) 0x0040001c54.0 -----DASM(fee7bc23)       [       356]
[==============================================================f===dp=i==========]-(         400000) 0x0040001c58.0 -----DASM(fed7e8e3)       [       357]
[==============================================================f===dp==i=========]-(         400000) 0x0040001c5c.0 -----DASM(00008067)       [       358]
[===============================================================f===dp===========]-(         400000) 0x0040001c60.0 -----DASM(00008067)       [       359]
[===============================================================f===dpi==========]-(         400000) 0x0040001c64.0 -----DASM(00c567b3)       [       360]
[===============================================================f===dp=i=========]-(         400000) 0x0040001c68.0 -----DASM(0077f793)       [       361]
[===============================================================f===dp===========]-(         400000) 0x0040001c6c.0 -----DASM(00c50633)       [       362]
[===============================================================f====dp==========]-(         400000) 0x0040001c70.0 -----DASM(0ff5f593)       [       363]
[===============================================================f====dp==========]-(         400000) 0x0040001c74.0 -----DASM(00078e63)       [       364]
[===============================================================f====dp==========]-(         400000) 0x0040001c78.0 -----DASM(00050793)       [       365]
[===============================================================f====dp==========]-(         400000) 0x0040001c7c.0 -----DASM(04c57263)       [       366]
[.cr....................................................................f..dpi...]-(         400000) 0x0040001c2c.0 DASM(00008067)            [       367]
[========================================================================f==dpi==]-(         400000) 0x0040001c40.0 -----DASM(fed57ee3)       [       368]
[========================================================================f==dpi==]-(         400000) 0x0040001c44.0 -----DASM(00050793)       [       369]

Pipeline Viewer shows that after 0x40001c40 there was a jump to 0x40001c2c with instruction 0x00008067, but in fact there was a jump to **0x40001c3c** with instruction 0x00008067.

ccelio, 2019-02-19:
RVC bugs... /shakes fist.

pentin-as, 2019-02-19:
I am testing a version without RVC. This: 378cbb5c7dbf5432c8d10bce9f5a959c3a99f1bf

jerryz123, 2019-02-19:
Sigh. The pipeline viewer is not as well maintained as it should be. Do you have the original trace which generated the pipeline visualization?

pentin-as, 2019-02-19:
367; O3PipeView:fetch:              471000:0x0040001c2c:0:       367:DASM(00008067)

pentin-as, 2019-02-19:
https://github.com/riscv-boom/riscv-boom/blob/2b1d7c7a3acb4165c4fb649954786e2ecdd20bee/src/main/scala/ifu/fetch-control-unit.scala#L647

chonsi, 2019-02-20:
Modify the [line 650](https://github.com/riscv-boom/riscv-boom/blob/2b1d7c7a3acb4165c4fb649954786e2ecdd20bee/src/main/scala/ifu/fetch-control-unit.scala#L650) to
`alignToFetchBoundary(bundle.pc) + (i << 2).U,`

It works for me!

chonsi, 2019-02-20:
BTW, my solution is only tested on earlier version withour RVC.
I am also woundering how to deal with the RVC support.

jerryz123, 2019-02-20:
Thanks @chonsi.
The PC for each instruction, accounting for RVC or misalignment, is calculated in F3 here: 
https://github.com/riscv-boom/riscv-boom/blob/2b1d7c7a3acb4165c4fb649954786e2ecdd20bee/src/main/scala/ifu/fetch-control-unit.scala#L302-L304

The O3 printout is generated from the results at the end of F3, so we can just use this vector of PCs instead of recalculating it from the fetchbundle pc.

jerryz123, 2019-02-23:
This should be fixed in #202

closed 2019-02-27 by jerryz123