riscv-boom/riscv-boom #216: MegaBOOM Compilation failure + BOOM rv64uc-v-rvc failure

akashsomeshwarrao, 2019-03-04:
Hello,

 Starting off with a disclaimer, I am new to the RISC V environment and the BOOM platform so please forgive the naivety of my questions/problems, if any.

I was installing and configuring the various configurations of BOOM from the BOOM-Template and ran into the following problems.

1. The MegaBoom config fails to compile with a java out of memory error. I changed the allocated heap size from 2Gb to 3Gb but then the compilation fails with an error 134 ( will get the entire log for details if required. It says something about a command failure) after running for a while.

2. While running the test suite on the default BOOM config the test "rv64uc-v-rvc” fails with the assertion "
first_pc: 0x0000005000 last_npc: 0x0000005002  Assertion failed: [fetchmonitor] A non-cfi instruction is followed by the wrong instruction. at fetchmonitor.scala:146 assert (first_pc === last_npc,”

3. I get the same assertion if I try running a simple hello world program on the same config

Note : I ran both the test and the hello world program on the SmallBOOM config and they both seem to work just fine.

Please let me know if anyone has any ideas on how to proceed.

Thanks and Regards,
Akash

jerryz123, 2019-03-04:
1. Can you show the log here?
2/3. Can you try this with the latest BOOM on master? Our regressions indicate these tests should pass.

akashsomeshwarrao, 2019-03-04:
**1.** make CONFIG=MegaBoomConfig run
java -Xmx3G -Xss8M -cp "/home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/firrtl/utils/bin/firrtl.jar":""/home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/target/scala-2.12/classes:/home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/chisel3/target/scala-2.12/*"" firrtl.Driver -i /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.fir -o /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.v -X verilog -faf /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.anno.json 
Total FIRRTL Compile Time: 2491981.2 ms
rm -rf /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig
mkdir -p /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig
/home/akash/Desktop/BOOM_RISC/boom-template/verisim/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 --no-threads -Wno-STMTDLY -Wno-CMPCONST --x-assign unique -I/home/akash/Desktop/BOOM_RISC/boom-template/testchipip/vsrc -I/home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/vsrc -O3 -CFLAGS " -O1 -std=c++11 -I/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/include -D__STDC_FORMAT_MACROS -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/verilator.h -include /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.plusArgs" -Mdir /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig \
-o /home/akash/Desktop/BOOM_RISC/boom-template/verisim/simulator-boom.system-MegaBoomConfig /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.v /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/SimDTM.cc /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/SimJTAG.cc /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/remote_bitbang.cc  /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/emulator.cc -LDFLAGS " -L/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/lib -Wl,-rpath,/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/lib -L/home/akash/Desktop/BOOM_RISC/boom-template/verisim -lfesvr -lpthread" \
-CFLAGS "-I/home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src -include /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig/VTestHarness.h"
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
%Error: Command Failed /home/akash/Desktop/BOOM_RISC/boom-template/verisim/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness \+define\+PRINTF_COND\=\$c\(\"verbose\"\,\"\&\&\"\,\"done_reset\"\) \+define\+STOP_COND\=\$c\(\"done_reset\"\) --assert --output-split 20000 --no-threads -Wno-STMTDLY -Wno-CMPCONST --x-assign unique -I/home/akash/Desktop/BOOM_RISC/boom-template/testchipip/vsrc -I/home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/vsrc -O3 -CFLAGS \ -O1\ -std\=c\+\+11\ -I/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/include\ -D__STDC_FORMAT_MACROS\ -DVERILATOR\ -DTEST_HARNESS\=VTestHarness\ -include\ /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/verilator.h\ -include\ /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.plusArgs -Mdir /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig -o /home/akash/Desktop/BOOM_RISC/boom-template/verisim/simulator-boom.system-MegaBoomConfig /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig.v /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/SimDTM.cc /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/SimJTAG.cc /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/remote_bitbang.cc /home/akash/Desktop/BOOM_RISC/boom-template/rocket-chip/src/main/resources/csrc/emulator.cc -LDFLAGS \ -L/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/lib\ -Wl\,-rpath\,/home/akash/Desktop/BOOM_RISC/RISCV_Toolchain/lib\ -L/home/akash/Desktop/BOOM_RISC/boom-template/verisim\ -lfesvr\ -lpthread -CFLAGS -I/home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src\ -include\ /home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig/VTestHarness.h
Makefile:70: recipe for target '/home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig/VTestHarness.mk' failed
make: *** [/home/akash/Desktop/BOOM_RISC/boom-template/verisim/generated-src/boom.system.TestHarness.MegaBoomConfig/VTestHarness.mk] Error 134

**2.** Im up to date with the master branch of BOOM. 
[image]

jerryz123, 2019-03-04:
1. How much memory does your machine have? Looks like you are running out of memory with bad_alloc.

2. You are on latest boom-template but not latest boom. Do 'cd boom && git pull origin master'

akashsomeshwarrao, 2019-03-04:
1. Im running this on a VM which has roughly 6GB allocated. With a java heap of 2GB, the compilation fails with a java main thread out of memory error. Got the above error with a heap of 3GB.

2. Thanks, looks like a couple of files are not up to date, though I pulled it just a week ago. Will run it again and provide an update.

jerryz123, 2019-03-04:
If you look closely, the error appears to be thrown by verilator, not Chisel or Firrtl

akashsomeshwarrao, 2019-03-05:
Points 2 and 3 are resolved with the up to date BOOM core thanks a ton.

Point 1 like you pointed out seems to be a memory issue coming from the Verilator , will try on a system with more memory I guess, though the makefile probably needs to be changed to have a 3GB java heap by default instead of the 2GB at present.

closed 2019-03-05 by akashsomeshwarrao