riscv-boom/riscv-boom #244: Add priority levels to asserts

jerryz123, 2019-04-16:
FireSim lets us synthesize all asserts or no asserts. Sometimes we want only some asserts.

Create a "boom_assert" macro which lets us specify levels for asserts.

For software simulation, we want all asserts.
For FPGA simulations, we want a subset of asserts which will fit on FPGA
For ASIC, we want a smaller subset of asserts for sanity checking.

sagark, 2019-04-16:
I think we should talk about a general way to do this for all designs being simulated, perhaps at firesim dev meeting today? @davidbiancolin

abejgonzalez, 2019-04-17:
@sagark @davidbiancolin any updates that boom should be aware of?

davidbiancolin, 2019-04-18:
No. I don't think these needs to be a macro though. I think it could be done pretty easily with a singleton that sets a debug level, you could change the level on a per-module or per-class basis that way (a la the FIRRTL logger). 

Unfortunately we can't annotate unnamed things (read: assertions) so you need to emit different assertions for different hosts.