riscv-boom/riscv-boom #249: `enablePrefetching` parameter is unused

jeehoonkang, 2019-04-18:
The `enablePrefetching` option in `BoomCoreParams` is not used at all.  Here is my grep result:

```
$ rg enablePrefetching
docs/sections/parameterization.rst
33:    enablePrefetching: Boolean = ?,

src/main/scala/common/parameters.scala
42:   enablePrefetching: Boolean = false,
```

I think this should somehow be documented, as is done for `enableAgePriorityIssue`:

```
src/main/scala/exu/issue-units/issue-units.scala
55:   require (enableAgePriorityIssue) // unordered is currently unsupported.
```

jeehoonkang, 2019-04-18:
Similarly, a few options in `HasBoomCoreParameters` are not used at all: e.g. `NUM_BROB_ENTRIES` and `BROB_ADDR_SZ`.

jerryz123, 2019-04-18:
Good call. We might add a prefetcher someday.

Some of those parameters are leftovers. We'll delete them soon. This is issue #167 

In general, the set of useful configuration parameters is those that are manipulated in `common/configs.scala`