riscv-boom/riscv-boom #265: HPM Questions

abejgonzalez, 2019-05-15:
Moving the question from #152 about HPMs to here 

Adapted from @heiner-bauer s question:

After reading chapter 12 of the documentation I have several questions regarding HPMs / HPEs:

- Which events does BOOM track? The number and ordering of the counters in BOOM does not seem to match the SiFive U54 events.
- How many independent counters does BOOM support?
- Do you happen to have some kind of library to use these counters (configuring and reading these CSRs is kind of fiddly)?

abejgonzalez, 2019-05-15:
Thanks for the question!

First of all, any suggestions to clarify the documentation is always helpful. Sorry it didn't help to much in this case.

1. Which events does BOOM track? 

Here are the snippets of code that define the events.

https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L190-L195
https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L221-L228
https://github.com/riscv-boom/riscv-boom/blob/f4968d770b6d0a461f81d26956a141e4300b3e45/src/main/scala/exu/core.scala#L245-L251

They tracks things such as branch prediction, ROB flushes, I/D$ misses, etc. As for the numbering/ordering of the events, we reference the U54 manual as an example on how to access events (if I remember correctly they map to the Rocket code). The events that they have (and what Rocket has) will be different from BOOM. You need to look at the `exu/core.scala` to see the signals that we chose and access them using the EventSet and EventBit that is relevant.

Note: If you want to track your own event, feel free to modify the source and add your own.

2. How many counters do we support?

We use the same infrastructure to support counters as Rocket. So if I am not mistaken we support 32 (including cycle, time, instret counters).

3. A library to use the counters (configuring and reading them is kinda ugly)?

What I end up using for my own projects is this repository: https://github.com/abejgonzalez/riscv-code-constructor. It includes a header file https://github.com/abejgonzalez/riscv-code-constructor/blob/master/inc/hpm-util.h and example on how to use it https://github.com/abejgonzalez/riscv-code-constructor/blob/master/src/hpmCounters.c. Otherwise, `riscv-tests` uses `setStats` to read the counters.

https://github.com/riscv/riscv-tests/blob/fbf5f3a2589c61d34569524dbf353beda0b6b4de/benchmarks/common/syscalls.c#L40-L54

Finally, Chris Celio created a program to help reading counters also: https://github.com/ccelio/riscv-hpmcounters

Otherwise, as far as I know, there is no "de-facto" standard for reading out counters. So you have to hack something together or use the code found in one of these sources (or somewhere else that I don't know of).

Please let me know if you have any other questions! I can also update the docs with this information if that was unclear.

heiner-bauer, 2019-05-21:
Thanks for the pointers to software examples.

So the mapping is like this ?

mhpeventX[7:0] = 0

| Bits  | Meaning |
| ------------- | ------------- |
| 8 |  Exception|

mhpeventX[7:0] = 1

| Bits  | Meaning |
| ------------- | ------------- |
| 8 |  I$ blocked |
| 9 |  - |
| 10|  Branch misprediction |
| 11 |  Control-flow misprediction |
| 12 |  Flush |
| 13 |  Branch resolved |

mhpeventX[7:0] = 2

| Bits  | Meaning |
| ------------- | ------------- |
| 8 |  I$ miss |
| 9 |  D$ miss |
| 10|  D$ release |
| 11 |  ITLB miss|
| 12 |  DTLB miss |
| 13 |  L2 TLB miss|

I assume that at unspecified bit positions( e.g.  with mhpeventX[7:0] = 1 at position 14) nothing happens, i.e. the counters will not increment. 
Why is it then necessary to have `"nop", () => false.B` ?

Regarding the number of counters, I just stumbled over `nPerfCounters` in Chapter 17, maybe this should be mentioned together with a more detailed description of the supported HPEs / HPMs ?

abejgonzalez, 2019-05-22:
Yup those mappings look right to me.

Flush goes high whenever the ROB is flushed due to things such as exceptions, pipeline replays, and mem. ordering failures. Its used to redirect the frontend of the pipeline through the Fetch Target Queue. For reference: https://github.com/riscv-boom/riscv-boom/blob/317b13756bdee864a8f84cad6258633273252201/src/main/scala/ifu/fetch-target-queue.scala#L174-L194

I$ blocked: https://github.com/riscv-boom/riscv-boom/blob/317b13756bdee864a8f84cad6258633273252201/src/main/scala/exu/core.scala#L258
It checks if there is a valid Fetch Packet coming from the Frontend.

As for the `nop` signals, with `false.B` they will not get incremented, correct. Why are they there? I don't exactly know, my guess is that they were put to original space out the BOOM HPMs so they matched Rocket, but over time both cores diverged and these were not updated. But as I mentioned earlier, feel free to override and change these (or if you find some useful signals to profile, put in a PR for the world to see!)

My main hesitation about adding a specific list of HPM/E's that BOOM support is that it could cause future divergence if the docs are not up to date. That's why I just point to the `exu/core.scala` file. However, I think a valid thing is to mention the number of counters we support.

abejgonzalez, 2019-05-28:
@heiner-bauer Any other questions on HPM's or is this issue resolved?

heiner-bauer, 2019-05-28:
Yes, you can close this issue. 
Thanks for your help !

abejgonzalez, 2019-05-28:
Great. Let us know if you have any more questions!

closed 2019-05-28 by abejgonzalez